BitMaker-hub/NerdMiner_v2 #219: Paswword is hardcodeed

PSLLSP, 2023-11-07:
Password is hardcoded to `x`. That is good default but advanced user should be able to configure password. For example, [zergpool](https://zergpool.com) uses password to define mining parameters, like miner ID, mined coin, etc. Other pools use password to define WorkerID. So, there are cases where user want to configure password...

I see that password is hardcoded, file [src/mining.cpp](https://github.com/BitMaker-hub/NerdMiner_v2/blob/master/src/mining.cpp)
```
      strcpy(mWorker.wName, Settings.BtcWallet);
      strcpy(mWorker.wPass, "x");
```

It means that new field has to be added to WiFi configuration form, to JSON configuration and to structure "Settings"...

PSLLSP, 2023-11-13:
> Today not really needed because current supported mining pools doesn't need it

This is a wrong assumption... Default password "x" is a good option but don't hardcode it and allow users to define their own password, it should be configurable. I assume you believe that only the only pool with support for NerdMiner is public-pool.io. Please, do not limit `NerdMiner` design to this limitation...

BitMaker-hub, 2023-11-29:
YOu can setup password on new update

PSLLSP, 2023-11-30:
I test firmware 1.6.3 with T-Display and I cannot confirm that this was fixed. Please, reopen the issue... I updated firmware with [WebFlasher](https://flasher.bitronics.store/)

1) I cannot see new field in WiFiManager, field for Pool Password
2) I tried to configure password with "curl" and doesn't take visible effect, I cannot see in the "test" pool that NerdMiner sends password

I am going to update "curl script" in issue #218 with `PoolPassword` field...

PSLLSP, 2023-12-02:
Configuration to test mining with "advanced" password. NerdMiner will mine `CHTA` in **solo** mode and there is a small change that block could be found, because this coin has low hashrate and could be mined with `cpuminer` or Android phone...

Mining [status](https://bowserlab.ddns.net:8080/?address=CSZfdb1ZPWT746iYkwtubLqStK1n5pL6k8), status shows NerdMiner version, password; good pool for troubleshooting stratum protocol...

```
{
  "SSID": "myNerdAP",
  "WifiPW": "mySECRET",
  "PoolUrl": "bowserlab.ddns.net",
  "PoolPort": 3333,
  "PoolPassword": "c=CHTA,m=solo",
  "BtcWallet": "CSZfdb1ZPWT746iYkwtubLqStK1n5pL6k8.TEST",
  "Timezone": 0,
  "SaveStats": false
}
```

The same config for `cpuminer`:

```
#!/bin/sh

# https://bowserlab.ddns.net:8080/
# https://bowserlab.ddns.net:8080/?address=CSZfdb1ZPWT746iYkwtubLqStK1n5pL6k8
# https://bowserlab.ddns.net:8080/api/walletEx?address=CSZfdb1ZPWT746iYkwtubLqStK1n5pL6k8

WORKER="TEST-CPU"
WALLET="CSZfdb1ZPWT746iYkwtubLqStK1n5pL6k8"

USER="$WALLET.$WORKER"
PASS="c=CHTA,m=solo"

OPTS="-t1"

cat <<EOF
REST API status https://bowserlab.ddns.net:8080/api/walletEx?address=$WALLET
$(curl -s "https://bowserlab.ddns.net:8080/api/walletEx?address=$WALLET")
---
EOF

./minerd --version
./minerd -a sha256d -o "$POOL" -u "$USER" -p "$PASS" $OPTS "$@"
```

Cr4zyT1zy, 2024-02-28:
I suggest this pool which works with nerdminer and SHA256.
https://unmineable.com/coins/BCH?algorithm=sha256
you can midify BCH with any crypto with SHA256 protocol.
Follow the instructions and if you want you can use my referral code
0ft1-s840

PSLLSP, 2024-02-28:
> I suggest this pool which works with nerdminer and SHA256. https://unmineable.com/coins/BCH?algorithm=sha256 you can midify BCH with any crypto with SHA256 protocol.

I like [`unmineable`](https://unmineable.com) pool. Unfortunately, I do not think it is good idea to mine in `unmineable` with `NerdMiner`. NerdMiner has really low has rate, it cannot compete with high end ASIC miners. Nerdminer is lottery miner, there is a believe it can find a block in **solo mining mode**. It is highly rare event but the probability of such event is not zero, so it is possible, at least in theory. It is game with rule "all or nothing".

**`Unmineable` pool doesn't support mining in solo mode**, that is the important point why I think it is not good idea to connect NerdMiner to `unmineable` pool. If you want really to mine in `unmineable`, set payout to `LTC` wallet, that coin has very low payout limit, just 0.00075 LTC. LTC is supported by many wallets, it has low payout limit at `unmineable`, low fees, it is well known coin, good coin to start your crypto experiments...  ;-)

I tried calculation, NerdMiner has hashrate about 75 kHps. The lowest unit that `Unmineable` has in the calculator for sha256 algo is **GHps** and I assume that 75 kHps is 0.000075 GHps (is this correct?). When I use 0.000075 GHps in the calculator, I see that such miner will earn NOTHING. There is no point to connect `NerdMiner` to `unmineable` pool...

PSLLSP, 2024-02-28:
> On Unmineable currency BCH I received 0.00000085 in these 20 hours.

OK. Minimal payout for BCH is set to 0.025 BCH by `Unmineable` pool. You earned 0.00000085 BCH in 20 hours. You will need 0.025 / 0.00000085 such sessions and  that is about 35294 days, only 96 years... ;-) Do you see the point? You can connect 100 NerdMiners, then you will mine 0.025 BCH in a year and you can ask for payout...  Try to mine LTC (Litecoin), it is easier to reach payout limit, I believe it is good advice...

Cr4zyT1zy, 2024-02-28:
Yes, in fact I connected it to this pool only because I wanted to see it work and certainly not to earn money from it.
As Doc Brown said in Back to the Future: It's a science experiment.
Have a nice day. :)

PSLLSP, 2024-02-28:
I love experiments... ;-) I agree that NerdMiner is an interesting experiment... And I think you still do not understand how `unmineable` works. The payout limit of the pool is important...

Current price for BCH is about 303.65 USD. Payout limit is 0.025 BCH, it is about 7.59 USD
Current price for LTC is about 74.96 USD. Payout limit is 0.00075 LTC, it is about 0.06 USD, just 6 cents!

You can ask for payout 135x sooner when you mine LTC than when you mine BCH.

You earned  0.00000085 BCH in 20 hours, that is about 0.000258 USD and you need to mine for 96 years to reach payout limit.
You can earn about 0.0000034 LTC in 20 hours (an estimate), that is about 0.000254864 USD  and reach payout limit in "just" 217 days; after 217 days you can ask for payout of 6 cents (based on current LTC price).

Or you can just mine in solo mode and risk that you will earn nothing (you will loose on hardware cost and electricity) or find a block that is worth of several thousand of dollars...

sany3001, 2024-02-28:
Good conversation guys, thanks for that!

PSLLSP, 2024-02-29:
I am on a business trip and I do not have any `NerdMiner` with me. I have a notebook and I tried to configure my notebook to mine `sha256d` in `unmineable` pool, I have hashrate about 8000 kHps, I use [pooler](https://github.com/pooler/cpuminer) CPU miner. I mine for more than 12 hours and I earned nothing (I am not surprised), I have not found any accepted share during this time period. Based on this experiment, I assume that `NerdMiner` that earned some BCH during 20 hours session was extremely lucky and that is not a repeatable result. Try to run `NerdMiner` connected to `unmineable` for a week, check the result, do a math and update this issue with your observation. I guess you will find that you need more time than those 96 years to reach payout limit for BCH...

This is my config for CPU miner (Linux):
```
$ cat mine-sha256-LTC-UM.sh
#!/bin/sh

WALLET="LTC:FIXME"  # replace FIXME with your LTC wallet
WORKER="TEST1"
REF="wvqn-jf6a"  # replace with your favorite REF code or leave empty, REF code decreases pool fee from 1.00% to 0.75%

POOL="stratum+tcp://sha256.unmineable.com:3333"
USER="$WALLET"
[ -n "$WORKER" ] && USER="$USER.$WORKER"
[ -n "$REF" ] && USER="$USER#$REF"
PASS="x"

OPTS="-t1"

./minerd --version
./minerd -a sha256d -o "$POOL" -u "$USER" -p "$PASS" $OPTS "$@"
```

PSLLSP, 2024-03-13:
I tried to mine with my NerdMiner in UnMineable pool and I can replicate it, my NerdMiner mines at 5 GHps... ;-) I assume there is a bug in the pool software

UnMineable pool sets difficulty to 16384:
```text
  Received message from pool
  Receiving: {"id": null, "method": "mining.set_difficulty", "params": [16384]}
    Parsing Method [SET DIFFICULTY]
    difficulty: 16384.000000000000
```
Then pool sends a "job" and just a second later pool updates difficulty, it change it to 0:
```text
  Received message from pool
  Receiving: {"id": null, "method": "mining.set_difficulty", "params": [0]}
    Parsing Method [SET DIFFICULTY]
    difficulty: 0.000000000000
```
At this point NerdMiner sends solution...
```text
   - Current diff share: 0.000017076509
   - Current pool diff : 0.000000000000
   - TX SHARE: c731e410d6b0fd4ed791e498d0dee6b9e3c3442102440007fbff0e16bfe40000
>>> Completed 0 share(s), 1974 Khashes, avg. hashrate 73.09 KH/s
```
And the solution is not rejected by the pool.

Maybe it is not a bug but a design, that some smart fox at UnMineable realized that it is better for them to pay just few cents a year to NerdMiner device. In the case that such NerdMiner device finds a block, the pool will earn a lot of money but miner will get just few cents...

PSLLSP, 2024-03-17:
Issue #352 created, it is about short username, I found that issue when I experiment with NANO at pool unMineable, see comment above for details.

Password field is hardcoded to "x" but other issue is that length of pool password field in code v1.6.3 is limited to 20 characters (19 in real, the last byte should be \0). 20 characters is too short, password field should be at least 100, 128 or even 256 characters...