BitMaker-hub/NerdMiner_v2 #220: Daylight saving time not supported

PSLLSP, 2023-11-07:
NerdMiner supports TimeZone in configuration, it is an integer. This is the first issue, there are timezones with offset of 30 or 45 minutes, etc. [strange timezones](https://www.timeanddate.com/time/time-zones-interesting.html)  To address  this issue, timezones, offset should be defined in minutes (60 means +1 hour offset)..

Other issue is, that timezone is defined as an offset from UTC. This works when timezone is not using [Daylight saving time](https://en.wikipedia.org/wiki/Daylight_saving_time). In other case, time is correct only for an half of the year; to fix the time, NerdMiner has to be reconfigured with new timezone every 6 months, like most of cheap clocks...

As most NerdMiner devices will never find BTC block, the clock is important function that gives this device a real usage, It will be nice to address time issue and define TimeZone in better way. With current NerdMiner, I would recommend  to run it in UTC timezone, that timezone doesn't have Daylight saving time and device will have correct time... ;-)

I checked code and I see that NerdMiner uses NTP protocol and reads time from NTP server `europe.pool.ntp.org`. It is hardcoded! It would be nice to have this as a configurable parameter... (file src/monitor.cpp)

Some hint how to address Daylight saving issue is described [here](https://forum.arduino.cc/t/ntp-daylight-saving-time-switching-problem/680916/2)

Other solution could be that a dedicated time server will be created for NerdMiner devices, it could be a mining pool service. Such time server will handle TimeZone information and will serve it to NerdMiner; NerdMiner just asks time server for a time and it will add a parameter "TZ" (timezone) and time server will serve local time. I do not understated why such feature was not already added to NTP (or SNTP) protocols, it will allow to design simple clients, really useful for IoT devices... (timezones are handled at server side, not by clients).

PSLLSP, 2023-11-22:
REST API for timezone, [WorldTime API](https://worldtimeapi.org/pages/examples); easy to use, returns correct time zone offset in minutes `raw_offset` and other useful information.

```
$ curl "https://worldtimeapi.org/api/timezone/Pacific/Kiritimati"
{"abbreviation":"+14","client_ip":"234.23.23.234","datetime":"2023-11-22T19:43:04.245664+14:00","day_of_week":3,"day_of_year":326,"dst":false,"dst_from":null,"dst_offset":0,"dst_until":null,"raw_offset":50400,"timezone":"Pacific/Kiritimati","unixtime":1700631784,"utc_datetime":"2023-11-22T05:43:04.245664+00:00","utc_offset":"+14:00","week_number":47}
```

When timezone is not configured in the device, timezone could be guessed on IP address.
```
$ curl "https://worldtimeapi.org/api/ip"
```

Really useful online service but not well known...

stl1988, 2024-05-25:
This is indeed a big problem. And currently, with firmware 1.6.3, you can't even change timezones anymore, it will always stay the one you defined at the first time setup.

tinychip0, 2025-01-03:
how to set time zone to +3:30