BitMaker-hub/NerdMiner_v2 #223: API to check miner status or configure mining pool

PSLLSP, 2023-11-07:
Nerdminer has WiFi connectivity and it has IP address. I can ping it, `ping` work. Unfortunately, I cannot find any API to check status of the miner, to check firmware version, uptime, hashrate, temperature, number of solved blocks, etc, etc

There is well defined API for miners, it is documented [here](https://github.com/ckolivas/cgminer/blob/master/API-README)

I think that implementing such API will open new possibilities for `NerdMiner`. It will be possible to run local monitoring server that will report status of each miner, to save data to log file, to change configuration from user friendly UI...

sany3001, 2023-11-08:
I would love to have this status too. It would be nice to have it as another screen on device or accessible via REST API or both.

Webranger2, 2023-11-11:
That would be great. then I could integrate that into the iobroker

dwightmulcahy, 2023-11-22:
I added #203 for something like this, too!

ChimpRocket, 2024-04-18:
+1 for this as well but it would be great if you return it using the some of the bitaxe api naming conventions in ap [Bitaxe](https://github.com/skot/ESP-Miner?tab=readme-ov-file#api), then we can easily include in the same monitoring. :-)  
Here's a sample

```
{
        "power":        13.359999656677246,
        "voltage":      4781.25,
        "current":      2795,
        "fanSpeed":     4503,
        "temp": 51,
        "hashRate":     655.23875991751333,
        "bestDiff":     "187M",
        "freeHeap":     177464,
        "coreVoltage":  1200,
        "coreVoltageActual":    1188,
        "frequency":    525,
        "ssid": "MySSID",
        "wifiStatus":   "Connected!",
        "sharesAccepted":       23563,
        "sharesRejected":       0,
        "uptimeSeconds":        287369,
        "ASICModel":    "Nerdminer?",
        "stratumURL":   "public-pool.io",
        "stratumPort":  21496,
        "stratumUser":  "bc1qy1111111111111111111111111111111111111.miner",
        "version":      "v1",
        "boardVersion": "1",
        "runningPartition":     "ota_0",
        "flipscreen":   1,
        "invertscreen": 0,
        "invertfanpolarity":    1,
        "autofanspeed": 0,
        "fanspeed":     60
}
```
Cheers

Thanks

b-rowan, 2024-07-27:
I am looking for this now as well, I have people asking about an integration with it.  Happy to provide support on API design if needed, but my C is not the strongest...

Reference issue - https://github.com/UpstreamData/pyasic/issues/176

toberkel, 2024-09-25:
Any updates on this? Would love to have a little API