BitMaker-hub/NerdMiner_v2 #224: T-Dongle-S3: issues

PSLLSP, 2023-11-07:
Issues with device `T-Dongle-S3`.

---

One button on device doesn't work as described in [README](https://github.com/BitMaker-hub/NerdMiner_v2/blob/master/README.md)

### One button devices:

- One click > change screen.
- Double click > change screen orientation.
- Tripple click > turn the screen off and on again.
- Hold 5 seconds > **reset the configurations and reboot** your NerdMiner. 

T-Dongle-S3 supports only "Double click" and "Hold 5 seconds". There is only one screen that scrolls all information, that is probably why "One click" is ignored. "Tripple click" doesn't work, maybe that hardware doesn't support screen on/off.

---

`config.json` file cannot  be read from microSDHC card.

T-Dongle-S3 has microSDHC slot, it is "hidden" in USB connector. [README](https://github.com/BitMaker-hub/NerdMiner_v2/blob/master/README.md) specifies how to prepare SD card with configuration to configure `NerdMiner`. I prepared 16 GB microSDHC card with `config.json` file but it seems like `NerdMiner` on T-Dongle-S3 ignores that file. It just shows `Use WiFi for setup...`; I tried to press button, to holed it, to hold button while I insert dongle to USB slot, etc.

My config.jso file is valid, it has 7 keys:
```
$ jq 'length' config.json 
7

$ cat config.json
{
  "SSID": "NerdAP",
  "WifiPW": "wifiSECRET",
  "PoolUrl": "solo.ckpool.org",
  "PoolPort": 3333,
  "BtcWallet": "bc1q28kkr5hk4gnqe3evma6runjrd2pvqyp8fpwfzu.NM0",
  "Timezone": 0,
  "SaveStats": false
}
```

lipe993, 2025-01-04:
Same... Does anyone else know what to do here?

sany3001, 2025-01-20:
You can try log on serial port - maybe there will be printed some useful info.

115200 | 8N1 for port setting if I recall well.