BitMaker-hub/NerdMiner_v2 #234: Device not mining / not getting block templates from mining pool

deyw-bit, 2023-11-21:
I have a new NerdMinerV2 device, which came preinstalled with everything setup. 
The device connects to WiFi, switching screen works fine, it displays correctly the current block, current time, correct global hashrate, etc.
BUT it doesn't mine. 
I have tried to change the configuration to various mining pools, but no success.
On the mining screen it shows: Block Templates 0, Best difficulty 0.0000 ... 0.00 KH/s.
I flashed it successfully with https://flasher.bitronics.store/ -- Successfully flashed NerdMinerv2 1.6.2 original board T-Display S3.
But the issue persists. It seems it just does not load block templates and thus is not mining. 
The device keeps restarting after 15 minutes.
What could be the issue? 

Console output:

Mounting File System...
mounted file system
reading config file
Opened configuration file
{ 
  "poolString": "public-pool.io",
  "portNumber": 21496,
  "btcString": "bc1<redacted>",
  "gmtZone": 1
}Parsing JSON
AllDone>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
[WORKER] ==> Mining subscribe
Sending  : {"id": 1, "method": "mining.subscribe", "params": ["NerdMinerV2"]}
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
Client not connected, trying to connect...
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
[WORKER] ==> Mining subscribe
Sending  : {"id": 1, "method": "mining.subscribe", "params": ["NerdMinerV2"]}
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
Client not connected, trying to connect...
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
[WORKER] ==> Mining subscribe
Sending  : {"id": 1, "method": "mining.subscribe", "params": ["NerdMinerV2"]}
...

The same log keeps filling up and the device restarts after 15 minutes.

PSLLSP, 2023-11-22:
It looks like your NerdMiner cannot connect to the pool. Try to replace `public-pool.io` in your config with IP address, `68.235.52.36`

```
$ host public-pool.io
public-pool.io has address 68.235.52.36
public-pool.io has IPv6 address 2607:9000:0:3:36a4:cf80:2f7b:6990
```

deyw-bit, 2023-11-23:
Thank you for your guidance! I attempted your suggestions and initially encountered some difficulties, but ultimately, they proved successful.

TL;DR: **It appears that my Internet Service Provider (ISP) is obstructing access to the mining pool via port 21496. I was able to circumvent this by using a VPN connection.**

Performed steps: 

As recommended, I updated the device configuration by replacing 'public-pool.io' with the IP address '68.235.52.36'. However, this modification did not resolve the issue.

I had a suspicion that my WiFi router might be obstructing the traffic for the mining pools. To address this, I adjusted the WAN settings on the router and allocated the NerdMiner device IP to the DMZ. This setup should theoretically permit any type of traffic to pass through without any filtration. This did not help.

Removed the DMZ settings on router, and instead enabled port forwarding from WAN to the NerdMiner device IP address for port 21496 for TCP. This did not help. 

Also on my modem enabled the Port-Forwarding WAN to router / port 21496 opened. This did not help. 

Enabled port forwarding for my Mac and as suggested, I have installed cpuminer and started the minerd: 

deyw$ /Users/deyw/Miner/minerd -a "sha256d" -o "stratum+tcp://public-pool.io:21496" -u "bc1..." -p "x" -t 1
[2023-11-22 18:14:17] Starting Stratum on stratum+tcp://public-pool.io:21496
[2023-11-22 18:14:17] 1 miner threads started, using 'sha256d' algorithm.
[2023-11-22 18:14:47] stratum_subscribe timed out
[2023-11-22 18:14:47] ...retry after 30 seconds
[2023-11-22 18:15:47] stratum_subscribe timed out
- this did not work. 

It appears that my internet service provider may be restricting access to the mining pool, specifically on port 21496. However, I can confirm that the provider isn't entirely blocking the site, as I am still able to access the website via web.public-pool.io.

I have started VPN on my Mac, and started the minerd: 

deyw$ /Users/deyw/Miner/minerd -a "sha256d" -o "stratum+tcp://public-pool.io:21496" -u "bc1..." -p "x" -t 1
[2023-11-22 18:21:28] 1 miner threads started, using 'sha256d' algorithm.
[2023-11-22 18:21:28] Starting Stratum on stratum+tcp://public-pool.io:21496
[2023-11-22 18:21:30] thread 0: 2097152 hashes, 8177 khash/s
[2023-11-22 18:22:09] thread 0: 338141656 hashes, 8584 khash/s
[2023-11-22 18:22:09] accepted: 1/1 (100.00%), 8584 khash/s (yay!!!)
[2023-11-22 18:22:28] thread 0: 172250048 hashes, 8962 khash/s
[2023-11-22 18:22:29] accepted: 2/2 (100.00%), 8962 khash/s (yay!!!)

Success! 

Solution: I have configured the NerdMiner device to connect to my iPhone via personal hotspot, activated the VPN on my iPhone, the device connected and started mining. After a minute it shows couple of received block templates, and it is mining at 55.40 KH/s.

An alternative approach could be to establish a continuous VPN connection for all devices directly through the router.

Website https://web.public-pool.io/#/app/bc1... shows the device stats / Your Best Difficulty 12.44 / worker 1 Sessions 55.3 KH/s

I plan to conduct further testing using an alternative ISP. It's quite frustrating that traffic is obstructed on port 21496. Perhaps there might be an alternate solution, aside from resorting to a VPN.

Anyway, now I have the ability to connect the NerdMiner to a power bank, and tether it to my iPhone's WiFi. This enables me to mine even while I'm on the go :-) It will be cool to bring this setup to the upcoming Bitcoin Meetup or ₿ conference. It's going to be a great conversation starter! 

I appreciate your advice! My big thanks also go to every single contributor who has dedicated their time and effort to this project! :-)

Closing this issue.

closed 2023-11-23 by deyw-bit

sany3001, 2023-11-23:
@deyw-bit May I ask for the name of your ISP? I want to put it on my black list. :-)

deyw-bit, 2023-11-23:
;-) 

[sany3001](https://github.com/sany3001) I am currently using vodafone.de as my ISP in Germany, connected through a 1GB cable line. However, I plan to transition to a fiber-optic line once another local ISP completes its network upgrade in my city. I am optimistic that after this, I will no longer have to contend with problems such as blocked ports.

PSLLSP, 2023-11-23:
`web.public-pool.io` is different server than `public-pool.io`; I assume that `web` is served by CloudFlare... It is difficult to say what is blocked

```
$ host web.public-pool.io
web.public-pool.io has address 188.114.97.9
web.public-pool.io has address 188.114.96.9
web.public-pool.io has IPv6 address 2a06:98c1:3121::9
web.public-pool.io has IPv6 address 2a06:98c1:3120::9

$ host public-pool.io
public-pool.io has address 68.235.52.36
public-pool.io has IPv6 address 2607:9000:0:3:36a4:cf80:2f7b:6990
```

I can ping the pool server:

```
$ ping -c1 public-pool.io
PING public-pool.io (68.235.52.36) 56(84) bytes of data.
64 bytes from 36.52.235.68.in-addr.arpa (68.235.52.36): icmp_seq=1 ttl=50 time=120 ms

--- public-pool.io ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 120.099/120.099/120.099/0.000 ms
```

sany3001, 2023-11-23:
I consider any blocking like something what any private company can do. I have no problem with this. But such a company will not get my money voluntary. So I will be avoiding Vodafone CZ too. :-)

PSLLSP, 2023-11-23:
IP address `68.235.52.36` was found in 3 blacklist: https://www.ipvoid.com/ip-blacklist-check/

Maybe that your ISP uses such lists to protect their customers...

sany3001, 2023-11-23:
I am far from expert level, but unidirectional blacklisting is not possible, right? Or can I send request and get response to resp. from IP and can the request from this IP still be blocked at the same time?
Like I can use services this IP offers, but this IP cannot "break" internet?

deyw-bit, 2023-11-24:
[sany3001](https://github.com/sany3001) & [PSLLSP](https://github.com/PSLLSP)

The actual problem at hand isn't related to my ISP blocking port 21496 - I delved into past issues and discovered a solution highlighted in this thread: https://github.com/BitMaker-hub/NerdMiner_v2/issues/180 

The actual obstacle was **my ASUS router which was blocking the port 21496**! The router has a feature called **AiProtection**, which was enabled. Once I disabled it, everything started functioning smoothly. 

This issue was originally identified and resolved in this discussion on reddit: https://www.reddit.com/r/MoneroMining/comments/m1g0y1/cannot_reach_any_pool_without_tls/

sany3001, 2023-11-25:
Glad to hear that Vodafone is not the problem. But staying watching this! :-)
Thanks for keeping us informed about final problem resolution!