BitMaker-hub/NerdMiner_v2 #236: esp32cam broken

valerio-vaccaro, 2023-11-24:
esp32cam board is broken due to

```
In file included from src/NerdMinerV2.ino.cpp:14:
src/drivers/storage/SDCard.h:30:14: fatal error: ..\lib\TFT_eSPI\User_Setup_Select.h: No such file or directory
```

BitMaker-hub, 2023-11-24:
let me take a look today at it.

jviki, 2023-12-20:
What do I do wrong? At Release-1.6.3, I get:
```
$ pio run -e esp32cam
...
In file included from src/NerdMinerV2.ino.cpp:14:0:
src/drivers/storage/SDCard.h:30:51: fatal error: ..\lib\TFT_eSPI\User_Setup_Select.h: No such file or directory

*******************************************************************************************
* Looking for ..\lib\TFT_eSPI\User_Setup_Select.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:..\lib\TFT_eSPI\User_Setup_Select.h"
* Web  > https://registry.platformio.org/search?q=header:..%5Clib%5CTFT_eSPI%5CUser_Setup_Select.h
*
*******************************************************************************************

compilation terminated.
Compiling .pio/build/esp32cam/src/drivers/displays/t_qtDisplayDriver.cpp.o
*** [.pio/build/esp32cam/src/NerdMinerV2.ino.cpp.o] Error 1
src/ShaTests/nerdSHA256.cpp: In function 'int Transform_Sha256(nerd_sha256*, const uint8_t*)':
src/ShaTests/nerdSHA256.cpp:120:5: error: 'IRAM_DATA_ATTR' was not declared in this scope
     IRAM_DATA_ATTR uint32_t W[NERD_BLOCK_SIZE/sizeof(uint32_t)];
     ^
src/ShaTests/nerdSHA256.cpp:76:20: error: 'W' was not declared in this scope
 #define SCHED1(j) (W[j] = *((uint32_t*)&data[j*sizeof(uint32_t)]))
                    ^
src/ShaTests/nerdSHA256.cpp:85:64: note: in expansion of macro 'SCHED1'
       t0 = h(j) + Sigma1(e(j)) + Ch(e(j), f(j), g(j)) + K[j] + SCHED1(j); \
                                                                ^
src/ShaTests/nerdSHA256.cpp:133:5: note: in expansion of macro 'RND1'
     RND1( 0); RND1( 1); RND1( 2); RND1( 3);
     ^
src/ShaTests/nerdSHA256.cpp: In function 'int nerd_midstate(nerd_sha256*, uint8_t*, uint32_t)':
src/ShaTests/nerdSHA256.cpp:233:9: warning: unused variable 'ret' [-Wunused-variable]
     int ret = 0;
         ^
src/ShaTests/nerdSHA256.cpp:234:14: warning: unused variable 'blocksLen' [-Wunused-variable]
     uint32_t blocksLen;
              ^
src/ShaTests/nerdSHA256.cpp:235:14: warning: unused variable 'local' [-Wunused-variable]
     uint8_t* local;
              ^
In file included from src/drivers/displays/dongleDisplayDriver.cpp:5:0:
lib/TFT_eSPI/TFT_eSPI.h:970:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
        ^
src/ShaTests/nerdSHA256.cpp: In function 'int nerd_double_sha(nerd_sha256*, uint8_t*, uint8_t*)':
src/ShaTests/nerdSHA256.cpp:351:5: error: 'IRAM_DATA_ATTR' was not declared in this scope
     IRAM_DATA_ATTR nerd_sha256 sha256;
     ^
src/ShaTests/nerdSHA256.cpp:358:5: error: 'sha256' was not declared in this scope
     sha256 = *midstate;
     ^
src/ShaTests/nerdSHA256.cpp:388:20: error: expected ';' before 'nerd_sha256'
     IRAM_DATA_ATTR nerd_sha256 secondSha256 = {
                    ^
src/ShaTests/nerdSHA256.cpp:353:9: warning: unused variable 'ret' [-Wunused-variable]
     int ret = 0;
         ^
src/ShaTests/nerdSHA256.cpp:354:14: warning: unused variable 'blocksLen' [-Wunused-variable]
     uint32_t blocksLen;
              ^
src/ShaTests/nerdSHA256.cpp:410:1: error: expected '}' at end of input
 }
 ^
src/ShaTests/nerdSHA256.cpp:410:1: warning: no return statement in function returning non-void [-Wreturn-type]
src/ShaTests/nerdSHA256plus.cpp: In function 'void nerd_mids(nerdSHA256_context*, uint8_t*)':
src/ShaTests/nerdSHA256plus.cpp:109:13: warning: unused variable 'i' [-Wunused-variable]
     uint8_t i;
             ^
*** [.pio/build/esp32cam/src/ShaTests/nerdSHA256.cpp.o] Error 1
src/ShaTests/nerdSHA256plus.cpp: In function 'bool nerd_sha256d(nerdSHA256_context*, uint8_t*, uint8_t*)':
src/ShaTests/nerdSHA256plus.cpp:207:13: warning: unused variable 'temp3' [-Wunused-variable]
     uint8_t temp3, temp4;
             ^
src/ShaTests/nerdSHA256plus.cpp:207:20: warning: unused variable 'temp4' [-Wunused-variable]
     uint8_t temp3, temp4;
                    ^
src/ShaTests/nerdSHA256plus.cpp:208:15: warning: unused variable 'buffer32' [-Wunused-variable]
     uint32_t* buffer32;
               ^
src/ShaTests/nerdSHA256plus.cpp:224:13: warning: unused variable 'i' [-Wunused-variable]
     uint8_t i;
             ^
src/drivers/displays/dongleDisplayDriver.cpp: In function 'void dongleDisplay_AlternateScreenState()':
src/drivers/displays/dongleDisplayDriver.cpp:106:34: error: 'TFT_BL' was not declared in this scope
   int screen_state = digitalRead(TFT_BL);
                                  ^
*** [.pio/build/esp32cam/src/drivers/displays/dongleDisplayDriver.cpp.o] Error 1
...
```