BitMaker-hub/NerdMiner_v2 #239: what happend - pool collapse?

TylonHH, 2023-11-26:
Hey there,
I noticed that my Nerdminer bootet in a loop and didnt get a WiFi connection. So I tried reflashing several times because it didnt flashed 1.6.2 until I found this was a [typo](https://github.com/BitMaker-hub/NerdMiner_v2/issues/209)

anyway. So I checked the pool (https://web.public-pool.io/#/) and saw this:
[image]

So it seems there was an global impact?! They restartet the pool? (Uptime: 1 day)
What happend
 
 I hope all the nerdminer user will notice that eventually ther gagdet are not running.

TylonHH, 2023-11-26:
LOL
[image]

BitMaker-hub, 2023-11-29:
Hi sorry, yes... was my typo error. Now you can check the 1.6.3