BitMaker-hub/NerdMiner_v2 #241: Firmware 1.6.3 fails on the LILYGO T-Display-AMOLED

drego85, 2023-11-29:
Hi Team,
I just tried to update my device firmware to version 1.6.3 via online flasher. 

The procedure is performed correctly and I get the message `Successfully flashed NMV2 1.6.3 on LILYGO T-Display-AMOLED` but the device does not boot, it remains black screen.

So I tried an `erase_flash` before doing the firmware write, from terminal but nothing changes.

I then tried writing firmware 1.6.3 from terminal and everything went well

```
python3 -m esptool --port /dev/cu.usbmodem21101 write_flash 0x0000 0x0000_bootloader.bin 0x8000 0x8000_partitions.bin 0x10000 0x10000_firmware_v1.6.3.bin 0xe000 0xe000_boot_app0.bin
```

So:
- If I update the firmware 1.6.3 via Web (https://flasher.bitronics.store/) it fails;
- If I update the firmware via terminal it's OK.

Ah another info the firmware 1.6.2 I installed it via Web, again with Chrome. And it worked perfectly.

BitMaker-hub, 2023-11-29:
Hi drego, thanks for your answer. 
Can you please try again with flasher?

drego85, 2023-11-29:
No, unfortunately still the same problem. 
NerdMinder does not start and the web portal instead shows that everything went well.

drego85, 2023-11-29:
I checked the hash of the binaries downloaded from the online flasher and they look the same as local files used to write the firmware.

```
https://flasher.bitronics.store/assets/nerdminer2_1.6.3_tdiplay_S3_Amoled/0x0000_bootloader.bin
https://flasher.bitronics.store/assets/nerdminer2_1.6.3_tdiplay_S3_Amoled/0x8000_partitions.bin
https://flasher.bitronics.store/assets/nerdminer2_1.6.3_tdiplay_S3_Amoled/0xe000_boot_app0.bin
https://flasher.bitronics.store/assets/nerdminer2_1.6.3_tdiplay_S3_Amoled/0x10000_firmware.bin
```

BitMaker-hub, 2023-11-29:
yes, really strange. Stupid question... Did you just tryed rebooting after flashing?

drego85, 2023-11-29:
Uhm, I tried to start the flash from an anonymous tab in Chrome and everything went well. Fucking Browser Cache! 😅

BitMaker-hub, 2023-11-29:
Yiiiha!

closed 2023-11-29 by BitMaker-hub