BitMaker-hub/NerdMiner_v2 #247: public-pool.io, not stable hashrate

PSLLSP, 2023-11-30:
`NerdMiner` connected to pool public-pool.io is not hashing at the highest possible speed, hashrate is not stable, it oscillates between zero and maximum. I was observing this issue for some time but I assume it is because miner is overheating or it has problem to connect to WiFi (too many WiFi devices in my place...).

When the same NerdMiner is connected to pool solo.ckpool.org, it hashes at stable hashrate, about 78 kHps (firmware v1.6.3).

I am not sure what is source of this issue, it could be firmware or pool. I bet on the pool, because I observed similar "unstable" behavior when I tried to mine other coins (BTH, XEC, CHTA, etc) in pools those are not optimized for NerdMiner low hashrate...

I test with NerdMiner v1.6.3 running on TTGO DISPLAY (ESP32-S3). I noticed this issue in the past with firmware v1.6.1 running on T-DONGLE-S3...

IT-Cru, 2023-12-01:
I have similar unstable rate at public-pool.io after upgrade NerdMiner with v1.6.3 release.

My 4 NerdMiners are connected via a QuatroBOARD 4 Nerdminers at a USB power connector which previously run stable at around 56kh/s on v1.6.2 release. Today I've upgrade one of them to v1.6.3 release for testing which show now this unstable hash rate. It is going up and down from 0 to 78kh/s.

When during flash of firmware (done via https://flasher.bitronics.store/) nothing in config has changed I would exclude issue from any WiFi connection, because other 3 ones are running stable with old 56kh/s hash rate with same WiFi connection on v1.6.2 release.

Let us now, if we could provide somehow any debug information for deeper investigation :)

julio-matarranz, 2023-12-01:
@IT-Cru are you sure you gave them enough power ? Maybe your board in 1.6.3 is possibly consuming more power ? My hashrate seem to be more stable after changing how I power my boards.

marssystems, 2023-12-01:
Power your NerdMiner with a 5V 2A mobile phone charger - it will run stable.

PSLLSP, 2023-12-01:
> Power your NerdMiner with a 5V 2A mobile phone charger - it will run stable.

No, that is not source of the issue. 5V/2A, that is only 10W. I power 3 NerdMiners  with USB hub that is powered with 15V/3A power supply, that is 45W and I still see those hashrate oscillations... That USB hub can can power 4 NewPak miners from GekkoScience and those run really hot, they require active cooling. NerdMiners doesn't require active cooling, I guess that they consume only about 1W each...  Power supply 5V/2A could power about 10 NerdMiners...

The best explanation of this mystery is that NerdMiner tries to send log data to serial port but PC is not listening so log data are stuck in the output buffer; read this [comment](https://github.com/BitMaker-hub/NerdMiner_v2/issues/247#issuecomment-1833436930)

IT-Cru, 2023-12-01:
Short update from my side.

I've tested to power up upgrade one without QuatroBOARD with same USB-C cable / power connection and then it runs stable at 78kh/s.

After this I reordered placements in positions on the board and when I put the already upgrade NerdMiner node at first slot it also runs stable at 78kh/s together with other non-upgraded ones.

My next step would be now to upgrade one more to see if it happens again.