BitMaker-hub/NerdMiner_v2 #248: Miner version is not sent to the mining pool

PSLLSP, 2023-11-30:
I updated my miner with firmware v1.6.3 but hashrate is still only about 55 kHps. This is for miner TTGO TDISPLAY v1.14. So, I am not sure if the miner was really updated to version 1.6.3 or if hashrate is "low" only because it has different CPU and it cannot hash at 78 kHps...

I cannot find any information about firmware flashed to the device, it is possible that [WebFlasher](https://flasher.bitronics.store/) updated device with older firmware, like v1.6.2 or v1.6.1. Unfortunately, I cannot see firmware version on display or on console, it is not send to the pool. Some pools can show miner version (Zergpool, MiningRigRentals), other could use it for statistical analysis or for troubleshooting...

I open this issue because I believe that miner should send firmware version to the mining pool. This is what I see in the serial console, only miner name is there, but no firmware version:

```
Sending  : {"id": 1, "method": "mining.subscribe", "params": ["NerdMinerV2"]}
```

This is [cpuminer](https://github.com/pooler/cpuminer), it sends version to the pool (use parameter `-P` to see protocol details):

```
[2023-11-30 00:02:23] > {"id": 1, "method": "mining.subscribe", "params": ["cpuminer/2.5.1"]}
```

Please, change parameter `mining.subscribe` call from `NerdMinerV2` to `NerdMinerV2/1.6.3`...

Please, print miner version to console during boot process, it is useful for troubleshooting....

BitMaker-hub, 2023-11-30:
Hi! Thanks for this information. I'll try what happens related to adding the version as a parameter on the public pool. 
I'll add version to log too

Regarding hashrate is exactly what you said, regarding adding ESP32 processor. ESP32S3 has IRAM and this brings NerdMiner faster response on CPU hashing functions

PSLLSP, 2023-12-10:
Miner information (version, password) can be see at this mining pool  [dashboard](https://bowserlab.ddns.net:8080/?address=1PKN98VN2z5gwSGZvGKS2bj8aADZBkyhkZ)

Configuration of NerdMiner to connect to that pool:
```
{
  "SSID": "myNerdAP",
  "WifiPW": "mySECRET",
  "PoolUrl": "bowserlab.ddns.net",
  "PoolPort": 3333,
  "PoolPassword": "c=BTC,m=solo,d=1",
  "BtcWallet": "1PKN98VN2z5gwSGZvGKS2bj8aADZBkyhkZ.TEST",
  "Timezone": 0,
  "SaveStats": false
}
```

When you cannot configure miner password (fw 1.6.3 doesn't support it), you can use just "x", it looks like pool remembers coin that was mined in the past for that address...