BitMaker-hub/NerdMiner_v2 #249: ERROR: *wm:[ERROR] parameter IDs can only contain alpha numeric chars

PSLLSP, 2023-11-30:
Device TTGO TDISPLAY v1.14 flashed with firmware 1.6.3 reports this error to the console during boot process:

```
*wm:[ERROR] parameter IDs can only contain alpha numeric chars
```

I am not sure what that means, like WiFIManager doesn't like some parameter... I am not sure about impact, device can mine... I report it because I do not like such mysterious errors... ;-)

Boot process details; note that there is no firmware version in this log, only `NerdMiner v2 starting...`:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
NerdMiner v2 starting......
SPIFS: Mounting File System...
SPIFS: Mounted
SPIFS: Loading config file
{
  "poolString": "public-pool.io",
  "portNumber": 21496,
  "poolPassword": "x",
  "btcString": "AddressWasHere.TTGO",
  "gmtZone": 0,
  "saveStatsToNVS": false
}
 *wm:[ERROR] parameter IDs can only contain alpha numeric chars 
AllDone: 
*wm:AutoConnect 
*wm:Connecting to SAVED AP: NerdAP
...
```

PSLLSP, 2023-11-30:
Error is related to password parameter, see [issue](https://github.com/BitMaker-hub/NerdMiner_v2/issues/219#issuecomment-1832932800)
New **Pool Password** parameter was added in v1.6.3 but it was not tested and it is not working, this error is related. Once parameter is fixed, error is gone...

thapwaritp, 2023-12-01:
[image]

BTC address not correct.

PSLLSP, 2023-12-01:
> BTC address not correct.

:-) No, address was "censored" for this post, that is not a issue... Check your device, with perfect address, error will be there too... ;-) The error is not related to BTC address field...

TV-Guy, 2024-04-20:
Using an ESP32-WROOM 30 or 38 pin device using a 135x240 TFT wired compatible with a TDisplay., I loaded  LILYGO TDisplay v1.14/0x10000_firmware.bin. 
Displays are shown OK.

But the Parameter IDs Error displays.

This is what was reported:

rst:Oxl (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) 
configsip: 0, SPIWP:Oxee clk_drv:Ox00,q_drv:Ox00,d_drv:Ox00,csO_drv:Ox00,hd_drv:Ox00,wp_drv:Ox00 ode:DIO, clock div:2 
load:Ox3fff0030,1en:1184 
oad:Ox40078000,1en:13192 
load:Ox40080400,1en:3028 
entry 0x400805e4 
NerdMiner v2 starting  
Button pressed to force start config mode 
SPIFS: Mounting File System... 
SPIFS: Mounted 
SPIFS: Loading config file 
{ 
    "poolString": "public-pool.lo",
    "portNumber": 21496, 
    "poolPassword": "x", 
    "btcString": "3DC352TjtQcggCMLuu39cozE9GyttGEUba",
    "gmtzone": -4, 
    "saveStatsToNVS": true
 } 
*wm:(ERROR] parameter IDs can only contain alpha numeric chars
AllDone:
*wm:StartAP with SSID: NerdMinerAP 
*wm:AP IP address: 192.168.4.1 
Entered Configuration Mode Config 
SSID: NerdMinerAP Config IP Address: 192.168.4.1 
*wm:Starting Web Portal 

As you can see, it errors, then kicks into web portal mode to allow configuration.

If I flash the ESP with the ESP32-DEVKIT BIN, it runs fine. 
Same Hardware including the display.
Also note the Password looks fine.
And the ESP32-DEVKIT bin doesn't report the button being pressed.