BitMaker-hub/NerdMiner_v2 #259: NerdMinerAP access point visible even all miners are connected to Wi-Fi and mining

sany3001, 2023-12-05:
I have connected 5 miners. All mining @~78kHs/s.
But now I see 2 devices started AP too. Observed using inSSIDer tool (Win).

All miners are running FW 1.6.3.

PSLLSP, 2023-12-07:
I have several NerdMiners running and I do not see `NerdMinerAP` SSID in the air. I experienced that one device lost configuration and entered WiFi configuration mode, I cannot explain it, maybe it overheated or some software error erased configuration. Maybe just a connectivity issue, that device was no able to connect to the mining pool so it switched to configuration mode... Is it possible that a temporary error at mining pool can erase NerdMiner configuration?? I will try to verify this idea... 

All your NerdMiners are mining and you still see `NerdMinerAP` in the air? Could you connect to it?? Then you can find MAC address at Info page. Maybe that your neighbor runs NerdMiners too .;-) Or some smart attacker just created a honeyspot with name `NerdMinerAP`... You have to try to find device that creates that AP, you can try to switch all your NerdMiners off and check that `NerdMinerAP` is not visible anymore...

sany3001, 2023-12-07:
I am (not now) running 5 miners LILYGO T-Display S3. All of them were mining when I have seen APs of two or three of them (it varies). All of them have quite big passive heat sink - I didn't noticed their were any kind hot.

However I can see that they are either rebooting or so as all values "total" (in the bottom right corner) were different.

When I have powered them off, all AP have gone. 

:-) I have just one neighbor and I would be very surprised if he would be running something like this. 

I am using this for those 5:
https://www.tindie.com/products/electronictrik/rack-1u-nerdminer/

PSLLSP, 2023-12-09:
I configured WiFi scanner on  my Linux box. First, I run [`horst`](https://www.dice.com/career-advice/monitor-wi-fi-signals-horst) for some time, then for several hours [`kismet`](https://www.kismetwireless.net/). I use WiFi dongle with chipset RT5370 and external antenna. There are many WiFi devices in this area, about 10 of them are different NerdMiners, mix of devices based on ESP32 and ESP32-S3. I have not captured any evidence, that `NerdMinerAP` is in the air, all my NerdMiner devices (firmware 1.6.3) are just connected to their WiFi AP and silently mine... `Kismet` only detected that one NerdMiner device is not immune against some WiFi attack and could be easily disconnected from AP by some magic WiFi packet...

PSLLSP, 2023-12-11:
My Kismet scanner found `NerdMinerAP`; it is a rare event, it took 3 days to capture such event. I am not sure what device is responsible for this event because MAC address of that AP is not associated with any of my devices (I did MAC address inventory during weekend). It is possible that new MAC address is assigned when NerdMiner is in WiFi configuration mode...

sany3001, 2023-12-12:
I can confirm that is is not permanent situation by me either. I am far from 24/7 monitoring, but today, I have not seen anything in the air yet.

I am only aware that this is feature of ESP32 that it can run AP and STA mode simultaneously. I do not have experience with ESP32 so I cannot add more here.

In my case all devices are also powered only (no data over USB).

PSLLSP, 2023-12-21:
I detected `NerdMinerAP` in the air today. I was no monitoring this issue anymore so I do not know how long NerdMinerAP is active. I was able to connect to `http://192.168.4.1/info` and found MAC address and client IP. I can ping the device on local WLAN address (like 192.168.132.167). I identified the device, it is `ESP-WROOM-32`. I checked the pool and I see it is mining. Device doesn't have display, it is "cheap" ESP32 module mining with hashrate about 55 kHps. It is not easy to identify what NerdMiner device is troublemaker; I created a database of MAC addresses of my devices, so I can find what device is troublemaker. It is not visible in any way... This case proves that this **issue is not limited to ESP32-S3 devices**. I have no idea how this issue was triggered.

sany3001, 2023-12-22:
By chance I have seen today also at least one `NerdMinerAP` Wi-Fi AP running. I was not investigating (but for sure it was LILIGO T-Display-S3. What I have observed was that at the same time 2 of my 5 miners (all same HW) were not mining (0 hashrate). Both were equipped with small external antenna (the others are not). Both devices were for some reason restarted recently (I didn't do that and there was no power outage). I will remove both antennas to have same HW for all.

I do know nothing about used FW, but I have the feeling that some WifiManager is used. I had some problems with this on another ESP8266 board in the past (mostly disconnections from Wi-Fi). Never solved, never really investigated at that time.

I will try to get more details. It is happening more often by me as I can see. (found it by chance when playing with phone)

What I know for sure is that all miners have weak Wi-Fi signal (hidden under steel). I can temporary move them closer to the router.

sany3001, 2023-12-22:
Not sure if this can be useful info, but I see (from inSSIDer) that when `NerdMinerAP` was active, it was each time on different channel (have seen 1,3,9).

Now all my devices are close to router. I shall have better control over them (especially restarts and hashrate drops).

randomn4m3, 2024-01-04:
I have noticed the same issue. Is it at least possible to change the password to connect to the NerdMinerAP?

PSLLSP, 2024-01-04:
> I have noticed the same issue. Is it at least possible to change the password to connect to the NerdMinerAP?

I think you can change WiFi password; you have to modify file `src/drivers/storage/storage.h` and build your customized binary/firmware... I think it will not affect QR code that is shown on the display because that is just an image...

BTW, what is your hardware? Is it T-Display-S3??

sany3001, 2024-01-19:
I have in my private project based on MicroPython (esp8266) did following "trick" after connecting my device to WiFi:
```
print("disabling AP mode")
ap_if = network.WLAN(network.AP_IF)
ap_if.active(False)
```
Maybe solution for our problem? I cannot test it yet (and for my project I need to make long term observations). 
I have not yet setup development environment for NerdMiner.
If someone will drop here link where to start it will be bonus for me.

PSLLSP, 2024-02-03:
My Ethernet switch in local LAN died yesterday. NerdMiners were able to connect to WiFi AP but were not able to connect to the internet or update IP address from DHCP server. My network was "broken"  for several hours, at least for 4 hours. I replaced the failed switch (power supply - AC adapter - died).
I see that many NerdMiners entered AP mode, there are many devices that mine and broadcast SSID `NerdMinerAP` after that incident. I assume that when miner cannot reach mining pool or it cannot refresh IP address from DHCP server, it activates AP mode `NerdMinerAP`... That could be a trigger for this issue. 

script to reset NerdMiners
```
for I in 123 124 125 211; do IP="192.168.10.$I"; echo "restarting $IP"; curl -sf "http://$IP/restart"; done
```

sany3001, 2024-02-03:
I have currently problem with building code for TTGO-T-Display (https://github.com/BitMaker-hub/NerdMiner_v2/issues/299) but as soon as I will be able to fix that I want to collect evidence for this topic and open ticket on https://github.com/tzapu/WiFiManager to discuss if this is WiFi Manager issue or not.
But to open ticket I need to try their last version, otherwise my ticket may be ignored.

sany3001, 2024-03-27:
Looks like this is not bug, but feature of WifiManager: https://github.com/tzapu/WiFiManager/issues/1725.
If I will find time, I will try to play with that.

bmaciag, 2024-11-01:
Any common solution till now to deactivate the AP?

sany3001, 2024-11-01:
I am still interested about solution of this problem. I cannot promise anything but I would love to open PR with solution proposal.

My opinion is that only user shall be allowed to activate AP mode (by long button press on T-display).