BitMaker-hub/NerdMiner_v2 #260: The new graphics in firmware 1.6.3 were not applied in the S3 Amoled board.

drego85, 2023-12-06:
Hello Team,
only yesterday I unfortunately realised that the new firmware 1.6.3 does not apply the new graphics (logo, background, BTC price, etc.) in the Lilygo T-Display S3 AMOLED board.

I'm certain that I have installed the latest version 1.6.3 because:
- At start-up it is indicated that firmware 1.6.3 is installed
- Hashrate is chastely above 70Kh/s

Tests performed:
- Online firmware flash via https://flasher.bitronics.store/
- Flash the firmware offline (download binaries from this repository)
- Several resets and erase_flash 

Unfortunately, I only have one Amoled board and cannot test with other boards.

Any suggestions are certainly appreciated.

Thanks

[image]

dackdel, 2024-01-30:
hey i can test with mine soon. but im having the trouble of not being able to connect to my router. got wpa3 password on a 5ghz network. can i ask what router is your device connected to?

drego85, 2024-01-30:
Hi, NerdMiner (ESP32) works only with 2.4Ghz WiFi Network!

dackdel, 2024-01-31:
ah okay. Thank you so much kind sir. Will fix that on the router right away.

dackdel, 2024-02-05:
hey so ive tested with both my amoled boards and do not see new graphics (or graphics that is different from what you have attached here) mine is the same as your image here. do you know what the new graphics looks like. Also I vote for a font change to inter https://github.com/rsms/inter its open source