BitMaker-hub/NerdMiner_v2 #261: Hash rate drops to 0 on 1.6.3; reboot loop

Z3tabyte, 2023-12-07:
I flashed 1.6.3 yesterday on Nerdminer V2 and noticed that every so often the hash rate would drop to zero for a few minutes, then recover.

Left overnight, the miner seemed to do okay. But this morning the hash rate dropped to zero and never recovered. About every 15 minutes the device would reboot and has never started hashing again.

Resetting the configuration has not worked.

PSLLSP, 2023-12-10:
Is your NerdMiner connected to active USB port? Then this issue could be a duplicate of #247

Z3tabyte, 2023-12-10:
No, it's connected to a USB hub, power only, no PC. Has been stable for the last 2 days 18 hours.

lucas3820, 2024-01-20:
I had this problem, I solved it by connecting it to a better cable and source, it was the power supply that wasn't adequate.

sany3001, 2024-01-20:
I am observing drops and reboots too. But not that often. It lasts sometimes couple of days or even more. 5x same HW, same power supply. I will try to catch if this is happening to only for some of miners or to all of them. If only to few of them, I would blame HW. I had one device that could be supplied from USB only, not via soldered pins. So pretty clear HW issue.

Mikhail0103, 2025-02-02:
I have the same problem and have already just tried everything, but without success! I also noticed that the public pool wasn't loading today either, and I assumed it was just a problem with the servers! Could it be?

У меня такая же проблема и уже просто все перепробывал но без успешно! Так же я заметил что общественный бассейн сегодня тоже не загружается и я предположил что просто проблемы с серверами! Может быть такое?

soenkegissel, 2025-02-08:
**Lilygo T-Display S3**
Had this problem too. Worked for a couple of hours, then 0.0kH and restarting all the time.
Flashed 1.6.4 **NerdminerV2** firmware is working with my Lilygo T-Display S3. I flashed with https://flasher.bitronics.store/
Seems to work now for some days already.