BitMaker-hub/NerdMiner_v2 #274: nerdminer_v2 is not staying connected to my pc/ screen wont work

raid3000, 2023-12-22:
so i received my nerd miner a few days ago and at first my screen wont turn on when i plug it in to my pc and when i was setting it up the screen went black and it wont come on anymore when i plug it just the little green light and red light turn on but no screen action. now when i plug it in to my laptop it wont stay connect i switch cords and still wont stay connected even when i connect it to my pc it does the same thing at this point am about to just toss it if it dose not work but before i do that am coming on here for some help or advice.

sany3001, 2023-12-22:
If I recall well, it is adviced to power it only without data. So either use power only cable to PC or use some other power supply like phone charger.

PSLLSP, 2023-12-30:
It could be a hardware issue. Or your firmware was not flashed.

First, try to [flash firmware](https://flasher.bitronics.store/).

Second, connect your NerdMiner to PC with data USB cable and run terminal for serial console, to check what happens inside your NerdMiner. On Linux, it can be done with command `miniterm /dev/ttyACM0 115200`. There are dozen ways how to do that on Windows, Windows is OS for beginners, so this task should be easy... ;-) Important point is that serial console of NerdMiner communicates at speed **115200 baud**...