BitMaker-hub/NerdMiner_v2 #279: Using OLED display with ESP32

victorlourenco-vhl, 2023-12-30:
Hey guys! Can anyone tell me how I can use the ESP32 WROOM with an OLED I2C 128x64 display?
[image]

19201003080114, 2024-02-14:
https://youtu.be/7XdNR2ou_-g

jeanfrodrigues, 2024-06-03:
this is a nice ideia.. i hope someone cam help!

paelzer, 2024-06-03:
Hi

That's easy, I have set this up here with VSCode & Platformio:
https://github.com/paelzer/NerdMiner_v2_paelzer

Display I2C pins:
SCL - ESP32 GPIO 22
SDA - ESP32 GPIO 21

Import the code into Platformio and select the env "ESP32_SSD1306".
Then compile and flash your board.

Should look like this:
[image]

Would be nice to know if it worked for you.

Boris

jeanfrodrigues, 2024-06-03:
very nice man, i'll try today and comeback to report, ths!

paelzer, 2024-06-03:
Sounds good. I will do some improvements next days and ask them to add it to the official NerdMiner_v2 repository.

Luis612207-radxa, 2024-10-28:
> Sounds good. I will do some improvements next days and ask them to add it to the official NerdMiner_v2 repository.

did they said yes? it would be very nice.

klenobi, 2024-12-29:
This is great! Thanks! Is it possible to update it to V2 to get up to 75khs? Currently getting 55khs on this code.

paelzer, 2024-12-30:
You need an S3 to get 75 kh/s. Not sure if an S2 is also capable, I don't have one so I can't test, but with a usual ESP32 it won't go over 58 something.

paelzer, 2024-12-30:
This one I tested, works fine at 78

https://www.waveshare.com/esp32-s3-zero.htm

klenobi, 2025-01-03:
OH! That makes sense. Thanks for making that clear, I am learning.

davidk85, 2025-01-24:
@paelzer Thanks for sharing... works with ESP32-S3 DevKitC-1 - only added other pins as build args