BitMaker-hub/NerdMiner_v2 #284: Stopping minicom makes the miner to stop working too

PsySc0rpi0n, 2024-01-03:
I have an ESP32 WROOM with no LCDs, screens, LiLos or whatever. Just the ESP32 WROOM alone, connected to a Raspberry Pi 4.
I access the ESP32 via ssh connection to the RPi4 and then via serial port.

Sometimes, when I have minicom running, the internet connection goes down and I have to close the ssh connection window.
When I start another ssh window to connect to the RPi 4 and then start minicom, the ESP32 WROOM seems to be rebooting.

This is what I see when I start minicom:
[image]

I presume this shouldn't happen. The firmware should be always running and not booting the ESP32 WROOM whenver a minicom session is closed for whatever reason!

Any thoughts?

PsySc0rpi0n, 2024-01-13:
Ok,  this post is about one other EPS-WROOM-32 I have at home, connected to a RockPro64 sbc. I also flashed it the same way I flashed the one I have at work, which is conencted to the Raspberry Pi 4.

I'll leave here some more info about my scneario at home. This ESP I have at home is one of those bought from China and uses an CP2102 UART. I connected it to a RockPro64 sbc (single board computer I have at home) but now I removed it for convenience and attached it to my laptop.

This is what I see in `dmesg`:
```
[76138.787194] usb 2-2: new full-speed USB device number 6 using xhci_hcd
[76138.966474] usb 2-2: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[76138.966492] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[76138.966499] usb 2-2: Product: CP2102 USB to UART Bridge Controller
[76138.966504] usb 2-2: Manufacturer: Silicon Labs
[76138.966508] usb 2-2: SerialNumber: 0001
[76139.004758] usbcore: registered new interface driver usbserial_generic
[76139.004776] usbserial: USB Serial support registered for generic
[76139.008690] usbcore: registered new interface driver cp210x
[76139.008710] usbserial: USB Serial support registered for cp210x
[76139.008744] cp210x 2-2:1.0: cp210x converter detected
[76139.009586] usb 2-2: cp210x converter now attached to ttyUSB0
```
However `lsusb` doesn't show it.
This is the command I use to flash the firmware binaries:
`esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 write_flash --verify --flash_freq 40m --flash_mode dio 0x10000 0x10000_firmware.bin 0x1000 0x1000_bootloader.bin 0x8000 0x8000_partitions.bin 0xe000 0xe000_boot_app0.bin`

Why you say I have the wrong firmware flashed?

I can try the cpuminer at my home board because I have heatsink and a fan on top of the cpu!

PSLLSP, 2024-01-13:
> Why you say I have the wrong firmware flashed?

I see errors in your serial console: 

```
[   138][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (132) gpio: gpio_set_level(226): GPIO output gpio_num error
[   141][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (140) gpio: gpio_set_level(226): GPIO output gpio_num error
```

The file `0x10000_firmware.bin` in the git repository is not the same as `0x10000_firmware.bin` used by Webflasher. You want to replace it with `0x10000_firmware_v1.6.3.bin` that is in the git repository. Or flash `NerdMiner` with [WebFlasher](https://flasher.bitronics.store/). I know that WebFlasher doesn't work for some ESP-WROOM-32 modules, I do not know why.

Check issue #266 for details...