BitMaker-hub/NerdMiner_v2 #285: ESP32-2432S028R wrong colors

sany3001, 2024-01-03:
I cannot find already reported problem with wrong colors on ESP32-2432S028R. I can confirm the same observation that was already reported.

[image]

paelzer, 2024-01-07:
Hello

Same problem here with wrong colors.
I use the ESP32-2432S028 version which has 2 USB ports (micro USB and USB C).
I tried with TFT_INVERSON_ON and TFT_INVERSON_OFF in USER_SETUP:h of TFT_eSPI library but these changes had no effect at all.
Does anybody know, where these changes need to be done so they have an effect?

BR
Boris

paelzer, 2024-01-08:
Hello again

I found this workaround which helped me solving my color issues: https://github.com/Bodmer/TFT_eSPI/issues/2985

By adding these lines below in the file **src/drivers/displays/esp23_2432s028r.cpp** into function **void esp32_2432S028R_Init(void)** inversion of colors was activated.

```
tft.writecommand(ILI9341_GAMMASET); //Gamma curve selected
tft.writedata(2);
delay(120);
tft.writecommand(ILI9341_GAMMASET); //Gamma curve selected
tft.writedata(1);
```

I added these lines under **tft.setRotation(1);** (in my case it was line )

[image]

Before it looked ugly like this:
[image]

Now it's nice:
[image]

Maybe this is helpful for somebody else as well.

Boris

BitMaker-hub, 2024-01-08:
Thanks Boris, yes this is the problem.
There is a color inversion. 

This is generally solved doing tft.swapbytes(1)

I'm sorry I couldn't get time for this, thanks for commenting it here.

sany3001, 2024-01-08:
I suggest to close this and keep open just https://github.com/BitMaker-hub/NerdMiner_v2/discussions/273.

closed 2024-01-09 by BitMaker-hub

RainBitcoins, 2024-01-28:
Could we just get this in a file release? How do I do this myself? In visual studio?

RainBitcoins, 2024-01-28:
Ok i got in and managed to change the values as per the instructions. I clicked save and reloaded 1.6.3 to my device but same thing. I know I'm missing out on a step....could someone explain it to me please? I managed to change it in Visual Studio but cant seem to make it work after that. Thanks to anyone.

sany3001, 2024-01-28:
Can we please continue this conversation in https://github.com/BitMaker-hub/NerdMiner_v2/discussions/273?

keukenrol, 2024-06-17:
> Hello
>
> Same problem here with wrong colors. I use the ESP32-2432S028 version which has 2 USB ports (micro USB and USB C). I tried with TFT_INVERSON_ON and TFT_INVERSON_OFF in USER_SETUP:h of [...]

Typo found here, for arduino IDE you need #define TFT_INVERSION_ON, you forgot the i in inversion :).

sany3001, 2024-06-19:
Works like a charm after changing configuration!

Thank you all who worked on this guys!

ArminXXXXX, 2025-09-27:
I am newbee and I can't find these file with arduino ide 🤔