BitMaker-hub/NerdMiner_v2 #286: Doesn't turn on when reset while using battery

orcunver, 2024-01-04:
Hello, sometimes I want to reset it when it drops to 0 kh (it usually happens after turning the screen on and off a few times, I think it's a bug). I have two lilygo t3 displays, both of them exhibit the same behavior. If I reset while using the battery, only the green LED turns on and the screen does not turn on. I have to connect the USB-C again and reset.

sany3001, 2024-01-06:
Not fully relevant comment - but I was observing drops to 0 kH/s when miners were having weak Wi-Fi signal. When I have moved them closer to the router, it looks much better looking at screens. I have never seen drop to 0.

BitMaker-hub, 2024-01-06:
Hi, I observed similar behaivour with last 1.6.3
I added some improvements for this on last updated changes to the repo. 

Check them and compile it whit them. This will solve your problem

closed 2024-01-06 by BitMaker-hub

orcunver, 2024-01-07:
Unfortunately, it didn't work, the same problem persists. I just did the update. I tried programming all the files from scratch, but the result was negative.