BitMaker-hub/NerdMiner_v2 #288: Using ESP32WROOM + ST7735 display

nicolocarcagni, 2024-01-06:
Hi there. simply the title... Is it possible?

SinexModelling, 2024-02-23:
I'm still trying :( but I couldn't find and produce a driver for st7735 1.8 inch. I have a st7735 1.8" 128 x 160 screen and a 38 pin esp32 wroom board. I only get the result from the cpp of the dongledisplaydriver. It has a 160 x 80 screen code and is currently working my esp32.. Maybe someone add st7735 160 X 128 image h.
[image]

matlen67, 2024-08-20:
Moin, i found this project for a week and build me a version. Its run on TFT 1.8 inch with ESP32-Wroom.

[image]

mathidios, 2024-11-18:
Hello my friends, help me by telling me how I should connect the wires so that this screen works on this ESP?
[image]