BitMaker-hub/NerdMiner_v2 #298: ESP32-S3-DevKitC-1U - possible to use with provided NerdMiner FW?

sany3001, 2024-01-19:
Hello there!
I would like to run NerdMiner FW on ESP32-S3-DevKitC-1U. I expect that it is possible. 
I have chosen ESP32-WROOM in web flasher. Flashed without problem.
However I do not see NerdMinerAP in the air. ;-(

julio-matarranz, 2024-01-19:
Dear,
Please check this thread : https://github.com/BitMaker-hub/NerdMiner_v2/discussions/294

sany3001, 2024-01-19:
Thanks for the reply. Already trying to build locally. Are there any plans to add this device to https://flasher.bitronics.store/ ?

sany3001, 2024-01-20:
Solved by building FW for `ESP32-S3-devKit1` in PlatformIO. Have to add minimalistic external Wi-Fi antenna to make it working.
Thanks for help @julio-matarranz and thanks for help [RedSalsicha](https://github.com/RedSalsicha)!

closed 2024-01-20 by sany3001

garunow, 2025-02-05:
My friends! which esp32s3 pins should the ili9341 screen be connected to?