BitMaker-hub/NerdMiner_v2 #299: master branch build is failing for TTGO-T-Display

sany3001, 2024-01-22:
I assume that TTGO-T-Display corresponds to LILYGO T-Display S3 HW (my physical device).

fatal error: sdkconfig.h: No such file or directory

sany3001, 2024-02-01:
I have tried even TAG Release-1.6.3 and I cannot compile from the same reason.
Please help.
I have deleted all code, made new Git clone and problem still exists. Nobody is building this option?

BitMaker-hub, 2024-02-01:
But are you buildig TTGO-Tdisplay S3 or the standard one?

sany3001, 2024-02-02:
Tested on TAG Release-1.6.3

**Not buildable:**
esp32cam
```
Building .pio/build/esp32cam/bootloader.bin
*** [.pio/build/esp32cam/src/NerdMinerV2.ino.cpp.o] Error 1
In file included from src/drivers/storage/SDCard.cpp:8:
src/drivers/storage/SDCard.h:30:14: fatal error: ..\lib\TFT_eSPI\User_Setup_Select.h: No such file or directory
```

TTGO-T-Display
```
Building in release mode
Compiling .pio/build/TTGO-T-Display/src/NerdMinerV2.ino.cpp.o
Compiling .pio/build/TTGO-T-Display/src/ShaTests/nerdSHA256.cpp.o
Compiling .pio/build/TTGO-T-Display/src/ShaTests/nerdSHA256plus.cpp.o
In file included from /home/jiri/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/sys/reent.h:503,
                 from /home/jiri/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/sys/reent.h:17,
                 from /home/jiri/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/stdio.h:60,
                 from /home/jiri/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:27,
                 from src/NerdMinerV2.ino.cpp:3:
/home/jiri/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/assert.h:20:10: fatal error: sdkconfig.h: No such file or directory
```

NerdminerV2-T-Display_V1
```
Building in release mode
Compiling .pio/build/NerdminerV2-T-Display_V1/src/drivers/displays/tDisplayV1Driver.cpp.o
In file included from src/drivers/displays/tDisplayV1Driver.cpp:5:
lib/TFT_eSPI/TFT_eSPI.h:970:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
        ^~~~~~~
Compiling .pio/build/NerdminerV2-T-Display_V1/lib2ac/FS/vfs_api.cpp.o
Compiling .pio/build/NerdminerV2-T-Display_V1/lib55d/SPI/SPI.cpp.o
Compiling .pio/build/NerdminerV2-T-Display_V1/libb92/SD/SD.cpp.o
Compiling .pio/build/NerdminerV2-T-Display_V1/libb92/SD/sd_diskio.cpp.o
Compiling .pio/build/NerdminerV2-T-Display_V1/libb92/SD/sd_diskio_crc.c.o
Compiling .pio/build/NerdminerV2-T-Display_V1/lib06d/SPIFFS/SPIFFS.cpp.o
src/drivers/displays/tDisplayV1Driver.cpp: In function 'void tDisplay_BTCprice(long unsigned int)':
src/drivers/displays/tDisplayV1Driver.cpp:203:67: error: 'priceScreen' was not declared in this scope
   background.pushImage(0, 0, priceScreenWidth, priceScreenHeight, priceScreen);
                                                                   ^~~~~~~~~~~
src/drivers/displays/tDisplayV1Driver.cpp:203:67: note: suggested alternative: 'price_screen'
   background.pushImage(0, 0, priceScreenWidth, priceScreenHeight, priceScreen);
                                                                   ^~~~~~~~~~~
                                                                   price_screen
Compiling .pio/build/NerdminerV2-T-Display_V1/liba49/OpenFontRender/FileSupport.cpp.o
*** [.pio/build/NerdminerV2-T-Display_V1/src/drivers/displays/tDisplayV1Driver.cpp.o] Error 1
```

**Buildable:**
ESP32-2432S028R, NerminerV2, ESP32-devKitv1, NerminerV2-S3-DONGLE, NerminerV2-S3-AMOLED,
NerminerV2-T-QT, M5-StampS3

sany3001, 2024-02-16:
I would like to contribute to this project, but when master branch is not buildable for me, it is hard. :-(
I have tried on another (Linux) computer with fresh clone and same problem for TTGO-T-Display.

```
 *  Executing task: platformio run --environment TTGO-T-Display 

Processing TTGO-T-Display (platform: espressif32; board: esp32dev; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- OpenFontRender @ 1.1.0+sha.f163cc6
|-- ArduinoJson @ 6.21.5
|-- WiFiManager @ 2.0.16-rc.2+sha.88c378f
|-- OneButton @ 2.5.0
|-- NTPClient @ 3.2.1
|-- WebServer @ 2.0.0
|-- WiFi @ 2.0.0
|-- TFT_eSPI @ 2.5.23
|-- rm67162
|-- SPI @ 2.0.0
|-- FS @ 2.0.0
|-- SD @ 2.0.0
|-- SD_MMC @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- HTTPClient @ 2.0.0
Building in release mode
Compiling .pio/build/TTGO-T-Display/src/NerdMinerV2.ino.cpp.o
Compiling .pio/build/TTGO-T-Display/src/ShaTests/nerdSHA256.cpp.o
Compiling .pio/build/TTGO-T-Display/src/ShaTests/nerdSHA256plus.cpp.o
In file included from /home/jiri/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/sys/reent.h:503,
                 from /home/jiri/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/sys/reent.h:17,
                 from /home/jiri/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/stdio.h:60,
                 from /home/jiri/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:27,
                 from src/NerdMinerV2.ino.cpp:3:
/home/jiri/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/assert.h:20:10: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://registry.platformio.org/search?q=header:sdkconfig.h
*
*******************************************************************

 #include <sdkconfig.h>
          ^~~~~~~~~~~~~
compilation terminated.
In file included from /home/jiri/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/sys/reent.h:503,
                 from /home/jiri/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/sys/reent.h:17,
                 from /home/jiri/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/stdio.h:60,
                 from src/ShaTests/nerdSHA256.cpp:2:
/home/jiri/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/assert.h:20:10: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://registry.platformio.org/search?q=header:sdkconfig.h
[... 12 lines cut ...]
*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://registry.platformio.org/search?q=header:sdkconfig.h
*
*******************************************************************

 #include <sdkconfig.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/TTGO-T-Display/src/NerdMinerV2.ino.cpp.o] Error 1
*** [.pio/build/TTGO-T-Display/src/ShaTests/nerdSHA256.cpp.o] Error 1
*** [.pio/build/TTGO-T-Display/src/ShaTests/nerdSHA256plus.cpp.o] Error 1
Compiling .pio/build/TTGO-T-Display/src/drivers/displays/amoledDisplayDriver.cpp.o
==================================================================================== [FAILED] Took 2.91 seconds ====================================================================================

Environment     Status    Duration
--------------  --------  ------------
TTGO-T-Display  FAILED    00:00:02.913
=============================================================================== 1 failed, 0 succeeded in 00:00:02.913 ===============================================================================

 *  The terminal process "platformio 'run', '--environment', 'TTGO-T-Display'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

julio-matarranz, 2024-03-13:
Dear,
The "sdkconfig.h" error is related to "board_build.arduino.memory_type = qio_opi".
You can remove/comment this line in plateformio.ini.

sany3001, 2024-03-14:
Thanks for you hint. In any case current master was buildable for all targets. Closing.

closed 2024-03-14 by sany3001