BitMaker-hub/NerdMiner_v2 #304: Suggestion : Screen auto off

Teppich80, 2024-01-27:
Hi,
Would be nice to have an auto-off-option for the screen.

It consumes around 1/3 of power and switches on when rebooted or longer connectivity loss.
You have to manully switch off again; auto-off e.g. after 30 seconds should be effective then.
Thanks

sany3001, 2024-01-27:
I would vote to make this configurable. As for me it is crucial feature that I can observe stats permanently.

Teppich80, 2024-01-27:
For sure. Should be optional. Checkbox or similar.

sany3001, 2024-02-06:
I would like to know if I can deactivate display of ESP32-2432S028R manually. Maybe not the best thread I should ask, but I will try here before opening new issue.

Teppich80, 2024-02-09:
> ESP32-2432S028R

It has to be managed by software. There is no button.

sany3001, 2024-02-09:
I have touch screen version.

BTW: suggestion can be open in "[Discussions](https://github.com/BitMaker-hub/NerdMiner_v2/discussions)". I have just found recently.