BitMaker-hub/NerdMiner_v2 #306: Bad graphics using ESP32-24325028

fcknzs-love, 2024-01-29:
After the first set up flawlessly, I got a second CYD. Unfortunately, with this one I have some display/graphics issues. The colors seem to be inverted, meaning black is white, white is black, yellow is blue and blue is orange. I also believe the image is slightly blurred.

[image]
[image]
[image]
[image]

And after all, this one doesn't start mining... CYD is from AliExpress (the cheepest, I could find...). Flashing only worked with https://flasher.bitronics.store . With ESP tool I got an MD5 mismatch..

[image]
[image]

paelzer, 2024-02-02:
Hello

Regarding the messed up colors...

Seems the firmware on https://flasher.bitronics.store/ is not optimized for the ESP32 2432S028 with two USB connectors. I have that one as well.

Try flashing with Platformio and select "ESP32_2432S028_2USB" as env.

[image]

Boris

UserCripto01, 2024-02-03:
Bom dia. Sou novo em programação e tenho o mesmo problema com minha nerminer.
Como faço para atualizar com a Platformio?

pedromfa, 2024-02-22:
> ESP32_2432S028_2USB

Hello. I have no idea how to work with Platformio.... Is there anyway to have the BIN file for this board and upload the firmware easier for a newbie here? Thank you. :)

sany3001, 2024-02-22:
It should be pretty easy to work with PlatformIO. Follow this guide:
https://platformio.org/install/ide?install=vscode

Then install Git command line tool. Not sure which one to advice. I am using https://git-scm.com/download/win under Windows.

Clone NerdMiner project
`git clone https://github.com/BitMaker-hub/NerdMiner_v2.git`

Open cloned folder in VScode.

Build with "button" at the bottom of page (choose target first - also on bottom of the page).

pedromfa, 2024-02-22:
Thank you so much, i think i got it. :)

pedromfa, 2024-02-22:
Btw, do you kwon how to turn off the screen without turning off the Miner?

pedromfa, 2024-02-22:
I only have 2 buttons on top. One says RST and the other say BOOT. Tried all possible combos but none just turns off the screen. This is the model with both USB C and micro usb ports.

sany3001, 2024-02-23:
> ![image](https://private-user-images.githubusercontent.com/158797249/307023313-ac284d91-6a07-407d-8c1f-cfb96775d8e9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

This guide is not applicable to this specific device.
RST button does reset of the device.

BOOT device works like on one button device:
[image]

(I have tried just single and double click)

More to this: you can somehow press screen to switch screens (if you have touch version).
Not sure how exactly shall this work. It is not very reliable to me.

djcasper1975, 2024-04-03:
thx i have now got a full colour screen if anybody wants the 2 usb screen ill give you the bins

djcasper1975, 2024-04-03:
[esp-2432S028_2USB.zip](https://github.com/BitMaker-hub/NerdMiner_v2/files/14855468/esp-2432S028_2USB.zip)
here is the fix i had no idea what im doing but follows 2 reply advice in here to dload software and clone etc and it worked! thankyou heres the final bins. if somebopdy can teach me how to get screen working on the other that would be epic ill do it right away and share it...

djcasper1975, 2024-04-03:
> > ESP32_2432S028_2USB
> 
> Hello. I have no idea how to work with Platformio.... Is there anyway to have the BIN file for this board and upload the firmware easier for a newbie here? Thank you. :)

@ yes i did what was said and made the files. 
[esp-2432S028_2USB.zip](https://github.com/BitMaker-hub/NerdMiner_v2/files/14855805/esp-2432S028_2USB.zip)

fluxres, 2024-04-09:
@djcasper1975 Thanks, the files fixed my screen issues.
What is the AP password ?  
MineYourCoins does not work.

Russty123, 2024-04-15:
What board did you choose when flashing? Mine is a single usb.  I chose the one ending in ‘R’ but screen doesnt work.

fluxres, 2024-04-15:
I ended up programming through platform IO.  I have the two USB variant and chose that in the build environment.  It worked a treat, have programmed 4 now with no issues.

Russty123, 2024-04-15:
Yeah im on the platformio. Ive tried the 2 usb build and the and the one ending in ‘r’  mine is just single usb and the board is esp32-2432s028.  When i load the 2usb build colour is pale and the image is off (see pic) and when i load the ‘r’ build screen is small. So which one build was the first one you tried before platformio? 
i might have to start another thread as another issue? 
[image]

Russty123, 2024-04-15:
Screen alignment is good when mining but colours are off. Or is yours still white? Blue and orange like this? I have other miners that look better than this. 

[image]

fluxres, 2024-04-15:
First time I used the "r" variant.  I had the same issues that are shown in your images.  The first time I used the online microminers flash tool as per git.  It's super easy.  Have you tried using that ?  It has worked fine for my "r" variant boards.

Russty123, 2024-04-15:
Ok i’ll give that a flash tool ago.  If it fails i found this edit of the code that i will try, its inverted colours. Thanks for your time and  help
[image]

IAmSixNine, 2024-04-26:
Thank You. I have the 2 usb one and had off colors. the zip file allowed me to fix this issue. Im not a programmer but trying to learn. So thank you for posting the files I needed.

mrijkhals, 2024-05-08:
I had this problem with some Arduino oled projects to, nothing to do with nerdminer.
The screens were not RGB but RBG so i had to change the library in order to get it work.
Maybe this fixed the issue over here to.... Anyway the new flash file worked fine thank you

BitMaker-hub, 2024-06-30:
Hi, this is solved on last version 1.7.0

closed 2024-06-30 by BitMaker-hub

RekOUF, 2024-11-02:
@sany3001 can u add esp32-2432s022n support also ? 

more info : 
https://github.com/lsdlsd88/2.2inch_ESP32-2432S022/tree/main/2.2inch_ESP32-2432S022
https://github.com/rzeldent/esp32-smartdisplay-demo/blob/main/platformio.ini

thanks

sany3001, 2024-11-02:
> @sany3001 can u add esp32-2432s022n support also ?
> 
> more info : https://github.com/lsdlsd88/2.2inch_ESP32-2432S022/tree/main/2.2inch_ESP32-2432S022 https://github.com/rzeldent/esp32-smartdisplay-demo/blob/main/platformio.ini
> 
> thanks

I am sorry, I am not active Nerd Miner developer. And it would cost me now quite a lot of time to doscover how to do that.

I hope that someone else will help.