BitMaker-hub/NerdMiner_v2 #309: ESPDUINO-32 doesn't mine when UART is not active

PSLLSP, 2024-02-03:
[ESPDUINO-32](http://www.esp32learning.com/hardware/a-look-at-the-espduino-32-board.php) with UART HL-340 (1a86:7523), flashed with NerdMiner firmware 1.6.3, flashed with WebFlasher. Configured to mine at vkbit.com mining pool. Device has ESP-WROOM-32 chip.

When it is connected to PC and serial console is active, ESPDUINO-32 starts to mine, blue LED blinks fast. no issue.

When it is connected to the same PC, with the same USB cable and USB port but serial terminal is not active, it doesn't mine. Blue LED is "ON" after reset and after about 50s it starts to blink slowly, like ESPDUINO-32 is in WiFi configuration mode. But I do not see NerMinerAP network in the air.

This is what I see in serial console (console has to be activated few seconds after reset, when blue LED is ON):
```
$ miniterm /dev/ttyUSB0 119200
--- Miniterm on /dev/ttyUSB0  119200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
*wm:AutoConnect: FAILED for  50523 ms
*wm:StartAP with SSID:  NerdMinerAP
*wm:AP IP address: 192.168.4.1
Entered Configuration Mode
Config SSID: NerdMinerAP
Config IP Address: 192.168.4.1
*wm:Starting Web Portal 
Failed to connect and hit timeout

Initiating tasks...
[MONITOR] started
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> Valid blocks: 0
>>> Block templates: 0
>>> Best difficulty: 0.000
[MINER]  0  Started runMiner Task!
>>> 32Bit shares: 0
>>> Temperature: 47
>>> Total MHashes: 0
[MINER]  1  Started runMiner Task!
>>> Time mining: 0  00:00:53
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>false / subscribed>false / wificonnected>false

[WORKER] Started. Running (Stratum) on core 1
TimeClient setup done
[Error] - current status: 6
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> Valid blocks: 0
>>> Block templates: 0
>>> Best difficulty: 0.000
>>> 32Bit shares: 0
```

I am not sure if this issue is there from the beginning and I missed it for 2 months or something was broken on the chip. I was doing a cleanup today, I have found that this device doesn't mine on the pool so I started to checking what is wrong. I re-flashed firmware but nothing was changed.

This is what I see when serial terminal is active and I press RESET; no issue, it mines:
```
$ miniterm /dev/ttyUSB0 119200
--- Miniterm on /dev/ttyUSB0  119200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
NerdMiner v2 starting......
No display driver initialized
Initializing...
SPIFS: Mounting File System...
SPIFS: Mounted
SPIFS: Loading config file
{
  "poolString": "pool.vkbit.com",
  "portNumber": 3333,
  "poolPassword": "x",
  "btcString": "btcWalletWasHere.NMDUINO",
  "gmtZone": 0,
  "saveStatsToNVS": false
}
*wm:[ERROR] parameter IDs can only contain alpha numeric chars 
AllDone: 
*wm:AutoConnect 
*wm:Connecting to SAVED AP: RouterAP
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.10.126

WiFi connected
IP address: 192.168.10.126
PoolString: pool.vkbit.com
portNumber: 3333
poolPassword: x
btcString: btcWalletWasHere.NMDUINO
TimeZone fromUTC: 0

Initiating tasks...
[MONITOR] started

[WORKER] Started. Running (Stratum) on core 1
Client not connected, trying to connect...
[MINER]  0  Started runMiner Task!
[MINER]  1  Started runMiner Task!
TimeClient setup done
CONNECTED - Current ip: 192.168.10.126Resolved DNS and save ip (first time) got: 167.114.0.221

TimeClient NTPupdateTime >>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> Valid blocks: 0
>>> Block templates: 0
>>> Best difficulty: 0.000
>>> 32Bit shares: 0
>>> Temperature: 47
>>> Total MHashes: 0
>>> Time mining: 0  00:00:03
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>false / subscribed>false / wificonnected>true
[WORKER] ==> Mining subscribe
  Sending  : {"id": 1, "method": "mining.subscribe", "params": ["NerdMinerV2"]}

  Receiving: {"id":1,"error":null,"result":[[["mining.notify","fdd2efea"]],"fdd2efea",4]}
    sub_details: fdd2efea
    extranonce1: fdd2efea
    extranonce2_size: 4
[WORKER] ==> Autorize work
  Sending  : {"params": ["btcWalletWasHere.NMDUINO", "x"], "id": 2, "method": "mining.authorize"}

  Sending  : {"id": 3, "method": "mining.suggest_difficulty", "params": [0.0001]}
  Received message from pool
  Receiving: {"id":2,"error":null,"result":true}
  Parsed JSON: Success
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> Valid blocks: 0
>>> Block templates: 0
>>> Best difficulty: 0.000
>>> 32Bit shares: 0
>>> Temperature: 48
>>> Total MHashes: 0
>>> Time mining: 0  00:00:04
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>true / wificonnected>true
  Received message from pool
  Receiving: {"id":null,"method":"mining.set_difficulty","params":[16384]}
    Parsing Method [SET DIFFICULTY]
    difficulty: 16384.000000000000
  Received message from pool
  Receiving: {"id":null,"method":"mining.notify","params":["286a8d","c8d2d1c00318a481e5e150bee9d0579b92c27d400001721a0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170350a50c5c766b6269742e636f6d5c","ffffffff034d3f53000000000016001427e04fe106a94b902e0f35ce87e0531b0ff47f33a534b640000000001976a914d66b6f65ce5b6b16d512e3ec2affe9eb5dbafbd888ac0000000000000000266a24aa21a9ed4f8771a6354f227590ea98272eef4f95fa13297559446eb5af59319feecf38f700000000",["3b238287119ca983751072f7e598117d2dc91bb57703f164cb5f4430cc360a90","5c297be613d055579e6e2cb45475da54c9b55cc04ed68e206ad22b9c2032528e","800741ea9939adbc5d6eaed6ad64987b8d65785826f4637623ff18dc078cd853","d80fc5f9c8d53978f78fc245ededc34dc0247493cb7715b4f9f10714b6e7f8a6","1b1187bda38278942d66b10ae95e6b07edfc6e8c5f417e58d7602da2af222b1a","b525cf9efa028d1e197e59213ab6dabcd34f6bd5734076cd1d0651a046b62032","d4a510a8a0ce6dbe5e6f26a175e4bad00ae8f054034dc6dab5c49700914efc45","60cf7ada055bd88d05a8aa69ffed2388ae6f85aeb6944079ba71434257bfbaf2","f30dc659441af87f74f92a36b73e1f112a046c26e52a5236d7184c09aa28ec47","99a37fa63a3247de8e99a26935a49ecb650df28899c4b85a9b47df2f9a998a99"],"20000000","1703ba5d","65be5ec1",false]}
    Parsing Method [MINING NOTIFY]
    target: 00000000000000000003ba5d0000000000000000000000000000000000000000
    coinbase: 02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170350a50c5c766b6269742e636f6d5cfdd2efea00000001ffffffff034d3f53000000000016001427e04fe106a94b902e0f35ce87e0531b0ff47f33a534b640000000001976a914d66b6f65ce5b6b16d512e3ec2affe9eb5dbafbd888ac0000000000000000266a24aa21a9ed4f8771a6354f227590ea98272eef4f95fa13297559446eb5af59319feecf38f700000000
    merkle sha         : ced31bf7d24fcee6c117019582d88782e25eb45d9b2ebede8298417bc8b80c50
  Received message from pool
  Receiving: {"id":3,"method":"mining.set_difficulty","params":[0.0001]}
    Parsing Method [SET DIFFICULTY]
    difficulty: 0.000100000000
>>> STARTING TO HASH NONCES
>>> STARTING TO HASH NONCES
>>> Completed 0 share(s), 39 Khashes, avg. hashrate 38.16 KH/s
>>> Valid blocks: 0
>>> Block templates: 1
>>> Best difficulty: 0.000
>>> 32Bit shares: 0
>>> Temperature: 48
>>> Total MHashes: 0
```

paelzer, 2024-02-03:
Hello hello, would you share pictures of your ESPDuinos front and backside and also the pinout that you use? Maybe it is similar to the one that I have, so I can give it a try as well?

Boris

PSLLSP, 2024-02-03:
I have this board (source AE):
[image]

I believe this is the same board, just another clone from other manufacturer...
[image]

paelzer, 2024-02-04:
Hello

I have tested with my board which is labeled as ESP32 D1 R32. I have similar issue. It works perfectly fine and is connected to a 5V power supply only. No serial connection.
Sometimes after reset it doesn't instantly starts mining. But later, after around ten minutes it starts mining, so I'm not sure if this is really a hardware issue. 

I flashed the board with PLATFORMIO using the Nerd Miner V2 master branch and I selected the ESP32 2432S028R as env.

Only one thing I changed compared to the default library setup for ESP32-2432S028 - I have setup IO22 as TFT_CS pin. By default IO15 is setup as TFT_CS pin but for this one is no socket soldered on the board. 

Actually I didn't check how this behaves with other boards I have here as I usually didn't really check if they had connection issues before your post. 

I will try this as well.

Boris

paelzer, 2024-02-04:
Btw this is the  board I used:

[image]

Boris