BitMaker-hub/NerdMiner_v2 #316: Alternative Pool to public-pool?

dapdahang, 2024-02-07:
With public-pool down now for more than 24 hours, are there good alternatives where I can also check my hashrate online? Ckpool apparently is not able to show the live stats.
Thanks guys

19201003080114, 2024-02-08:
stratum+tcp://pool.vkbit.com:3333

sany3001, 2024-02-08:
Have you predicted future @dapdahang? For me went https://web.public-pool.io/ down just some minutes ago local CET time.

dapdahang, 2024-02-08:
Yeah, while it was running for me today again, it went down again like 20
minutes ago.
I am now trying vkbit, so far looking good

jorgb90, 2024-02-08:
for me its not saving pool address when in config mode. It always goes back to airdns..

KaSt, 2024-02-12:
I just opened mine, after public-pool's semi downtime days ago and I also wanted to have mine. Using  public-pool's source code. In case the Developer is interested and agrees in me posting it here, I shall do it. Still in test, checking stability and at the moment only my miners (4) are on it.
3 NerdMiners running on:
- 2 LilyGo Display 1.4
- 1 LilyGo QT-Pro

Also have:
- 1 PIZero with four old USB miners, for a total of around 3Gh/s.

Good old days when I could mine with the CPU of my computer.

mynameismichele, 2024-02-27:
I also tried running my own [public-pool](https://github.com/benjamin-wilson/public-pool) on umbrel and it is working fine so far.
It's a pity that the total hashrate showed on ESP32-2432S028R is not working anymore because I think it was connected only to public-pool.io , any chance we can get it back @nitroxgas?