BitMaker-hub/NerdMiner_v2 #326: web.public-pool.io is down

valshebnik, 2024-02-14:
Issue is with the mining pool https://web.public-pool.io/, it is "down"

Miners has all zeros.
[image]

The site https://web.public-pool.io doesn't show the info. It takes forever to load content.

Because web is not accessible, I cannot find contact information to admins, I cannot read "news", etc. I hope that someone in this community can contact administrators of the pool and ask them to address the issue...

19201003080114, 2024-02-14:
Public pool down is NOT a Nerdminer issue 🤦‍♂️

sany3001, 2024-02-15:
I am sure, you will hate me, but because mentioned pool is part of current FW of NerdMiner, it IS problem for this project. Root cause is different, but we need to act in this project too.

sany3001, 2024-02-15:
> Issue is with the mining pool https://web.public-pool.io/, it is "down"
> 
> Miners has all zeros. ![photo_2024-02-14_13-00-55](https://private-user-images.githubusercontent.com/1078393/304748790-be7ef5ea-2fd6-4349-be03-c2fa9b62324e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgwMDQ3MzAsIm5iZiI6MTcwODAwNDQzMCwicGF0aCI6Ii8xMDc4MzkzLzMwNDc0ODc5MC1iZTdlZjVlYS0yZmQ2LTQzNDktYmUwMy1jMmZhOWI2MjMyNGUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTVUMTM0MDMwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODU5ZjNjMTFlMmNhODJmZDcyM2IxMjAzNTA2ODFhY2NjMmI3MDE3YTUyYjJlNTA0ZDQzZjljYWRmNWJmZjFjNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.kqhEL8pTOyVp8qP4toTHB5j89tg5C4kCqKR28zLPXEE)
> 
> The site https://web.public-pool.io doesn't show the info. It takes forever to load content.
> 
> Because web is not accessible, I cannot find contact information to admins, I cannot read "news", etc. I hope that someone in this community can contact administrators of the pool and ask them to address the issue...

Plese open ticket here too: https://github.com/benjamin-wilson/public-pool

sany3001, 2024-02-15:
Could be that this problem is somewhere else. like geolocation, local provider ... because when it is reported here that pool is down, it is working for me. And it was not working for me during another period of that day for me.

19201003080114, 2024-02-15:
Pre-filling public pool in the AP setup screen IMO is not good programming practice. 
When (not if) public pool is down, Nerdminer community is flooded with people complaining connection issue when it has nothing to do with Nerdminer. 
it’s like when your local cell service is down, and you complain to Apple/Google/Samsung/etc…they have nothing to do with that.

sany3001, 2024-02-15:
> Pre-filling public pool in the AP setup screen IMO is not good programming practice. When (not if) public pool is down, Nerdminer community is flooded with people complaining connection issue when [...]

I understand and agree.

However it is long path to the pool. Anything in the path can get broken.
In such or similar error state could NerdMiner clearly write error message on screen like:
`no access to provided pool (check your network and check if the pool is accessible from your network)`.
Or any even better message. Otherwise NerdMiner will always be blamed by someone as not working.

sany3001, 2024-02-15:
@valshebnik, if you agree, close this issue - you can continue here: https://github.com/BitMaker-hub/NerdMiner_v2/discussions/327

closed 2024-02-15 by valshebnik

00impera, 2025-08-01:
Publicpool its down?

BitMaker-hub, 2025-08-01:
public pool is currently down, let them some time to upload it again or use a 2nd pool