BitMaker-hub/NerdMiner_v2 #330: How to Acess Nerd Miner after configured ?

LuanFMelo, 2024-02-21:
I cant access nerdminer by its ip and port at the browser and cant find any settings that can help me with

19201003080114, 2024-02-22:
That is normal.
Once the settings are saved, AP is closed and there is no web server to show settings.
To view the settings, you need to press Boot and power up by USB (that means you need to power cycle).
After power cycle, AP will be active and you can login to view settings.
To erase settings, you need to hold Boot for 5+ seconds. The unit will reboot.
After reboot, AP will be active and all settings are reset to factory default. You can login to view and change settings.

closed 2024-03-02 by BitMaker-hub

Blackhawk17-USA, 2025-02-25:
Is there a way to erase all previous data history completely? I moved to a different pool and would like to see only the new data on the NerdMiner v2 going forward?  I'm not talking about resetting the configuration, in other words, I want to erase all data like the miner was the first day I first received it in the mail. :)

MyOwn2C, 2025-02-26:
Use ESP tool to wipe the flash. Then you can reprogram it again

Blackhawk17-USA, 2025-02-26:
I was finally able to erase all the data/history on my NerdMinerV2 
1. Maybe it was holding the Top Button and the Bottom Button at the same time.
2. Maybe it was holding the Centre Button.
3. Maybe it was deselecting the Track Uptime, Best Diff, Total Hashes in device Flash memory. (Experimental)
I think it was number 3.  Now all the old data was finally removed and it was just like it was when I opened it out of the cardboard box from the store. LOL Please message me back if you know the solo pool that will give the best chance of winning a block or is it better to create a home bitcoin node and runt it from there?  I know the chances are extremely low of winning a bitcoin block with a NerdMiner but either way what is the best possible way to try? Thank you!