BitMaker-hub/NerdMiner_v2 #331: NerdMiner runs out of KH/s

ricardonegrao, 2024-02-22:
This problem is occurring and I have already reset the equipment several times.

Has anyone ever experienced this?

Is there a solution?

[image]

19201003080114, 2024-02-22:
Either wrong settings, or you try to connect to 5G wifi. 
Nerdminer only works on 2.4G WiFi.

ricardonegrao, 2024-02-22:
I am using 2G.

How would you know what is happening with this equipment?

19201003080114, 2024-02-22:
Rest it and check your pool and port settings

ricardonegrao, 2024-02-22:
This is my pool configuration:

Pool url pool.vkbit.com
Pool port 3333
-3

19201003080114, 2024-02-22:
Try resetting your wifi router.

DanielErnany, 2024-03-02:
> I found the problem:
> 
> It was my BTC wallet address!
> 
> Thank you for your help.

What was wrong with her? and how did you fix it? I have the same problem

v1per4llbl4ck, 2024-03-02:
me the same i dont know how to change btc address. with sdcard dont work. i take my address from crypto.com in receve mode

19201003080114, 2024-03-04:
> me the same i dont know how to change btc address. with sdcard dont work. i take my address from crypto.com in receve mode

Boot to Wifi Manger and you can change the btc address

19201003080114, 2024-03-04:
> > I found the problem:
> > It was my BTC wallet address!
> > Thank you for your help.
> 
> What was wrong with her? and how did you fix it? I have the same problem

Boot to Wifi Manger and you can change the btc address

MARK345456, 2024-03-07:
ALGUEM SABE COMO REINICIAR O NERDMINER

dwightmulcahy, 2024-03-13:
If you see your Nerdminer go to 0.00kH/s a lot, it could be the cable or power source you are using.  I noticed that it bounces between 78kH/s and 0.00kH/s when I have it hooked up to my laptop, but when I plug it into a 1 amp or better power supply, it stays at the 78kH/s-ish mining all the time.

ricardonegrao, 2024-03-13:
https://flasher.bitronics.store/

ben6608, 2024-03-24:
I also have the same problem. I updated the BTC wallet address and it didn't improve. I also changed the pool and reset the flash machine, but it didn't work.

dwightmulcahy, 2024-03-24:
Have you tried a different power supply and cable to eliminate that?

ben6608, 2024-03-24:
Yes, I tried changing the power supply and cables, but it didn’t work.

19201003080114, 2024-03-26:
> Yes, I tried changing the power supply and cables, but it didn’t work.

Check your settings:
- It can only connect to 2.4 Ghz wifi
- Make sure SSID and password are correct
- Make sure pool address, pool port and BTC wallet address are correct

Loickrypto, 2024-08-17:
same issue . the problem was the btc adress . works well now

Blackhawk17-USA, 2025-02-21:
1. I use Public Pool so this is my Pool Config stratum+tcp://public-pool.io:21496
2. My BTC Address has always been good
3. I'm connected to the correct WiFi at 2.4 Ghz
4. I successfully flashed NerdMinerv2 to 1.6.3 original board T-Display S3
5. I removed USB cable from computer and plugged into 5 Watt USB adapter  👍👍👍 
Seems like the only thing that fixed it, was number 5, the POWER ADAPTER upgrade. LOL
Thank you Dwight Mulcahy 💯

andrewlimaza, 2025-10-20:
My WIFI SSID had spaces in the name. I created a new guest network with no spaces "guestnetwork" and now it connects 100% of the time. Hope this helps anyone.