BitMaker-hub/NerdMiner_v2 #336: Need some help updating nerdminer v1.6.3 to my Lilygo S3

GFreedomX, 2024-03-01:
I flashed my S3 using both nerdminerv2 flash and the ESP-tool
but my unit is giving me a black screen and when I plug it into my desktop it keep looping ( I think cause I keep hearing the USB device connect and disconnect)

wonder if anyone have this issue?

19201003080114, 2024-03-01:
Get to bootloader mode before flashing

GFreedomX, 2024-03-01:
i am in the bootloader mode when I did the flashing

correct me if I am wrong
I will need to hold done the boot button and power it on to get into the bootloader mode

19201003080114, 2024-03-01:
[image]

GFreedomX, 2024-03-02:
[image]
i am able to upload but i am not sure what is the issue with my unit.

is the leaving part normal?

my unit is not able to boot up 
kind of puzzle about this

19201003080114, 2024-03-02:
After seeing “Leaving…”, do a power cycle and you will see QR to access WiFi manager.

GFreedomX, 2024-03-02:
damn let me go try it

GFreedomX, 2024-03-02:
its working
a typo cause all this issue.
very sorry for the trouble cause

closed 2024-03-02 by BitMaker-hub