BitMaker-hub/NerdMiner_v2 #357: Double NerMiner

pmiguelr, 2024-03-24:
How I can build one double with 2 S3 displays?
I need to by it soldered with headers?

sany3001, 2024-03-24:
Something like this?

https://www.tindie.com/products/electronictrik/quatrominer-for-nerdminer/

pmiguelr, 2024-03-24:
Yes. Both for two?And in code. What I need to do?Where I can find shield for two?Board is with header soldered?Enviado de Outlook para AndroidDe: sany3001 ***@***.***>Enviado: domingo, março 24, 2024 7:19:05 da tardePara: BitMaker-hub/NerdMiner_v2 ***@***.***>Cc: pmiguelr ***@***.***>; Author ***@***.***>Assunto: Re: [BitMaker-hub/NerdMiner_v2] Double NerdMiner (Issue #357)
Something like this?
https://www.tindie.com/products/electronictrik/quatrominer-for-nerdminer/

sany3001, 2024-03-27:
This is just HW. No need to do any changes in SW. All devices are running independently.
If you setup same BTC address in all your devices you can see statistic on device like ESP32-2432S028R. See below.

[
[image]
](url)

[image]