BitMaker-hub/NerdMiner_v2 #369: "configuration mode" not working for ESP32-WROOM-32

ChicoCybergoth, 2024-04-05:
I'm trying to use this function:
"Hold and power up > enter configuration mode and edit current config via Wifi. You could change your settings or verify them."

But without success, I've tried every possible way, NerdMiner always boots in normal function.
Board ESP32-WROOM-32.
V1.6.3 (bin ESP32-devKit)

19201003080114, 2024-04-05:
> I'm trying to use this function: "Hold and power up > enter configuration mode and edit current config via Wifi. You could change your settings or verify them."
>
> But without success, I've tried [...]

You need to disconnect power, then hold the button while inserting power again.

ChicoCybergoth, 2024-04-05:
@19201003080114 yes, I have tried several times in several ways, always boots in normal mode.
I believe it is not working.

sany3001, 2024-04-06:
Could it be there is no solution for one button device (I consider wroom to be such a device - reset button is not regular button, if I am right).

[image]

My solution for similar situation was reset configuration / reflash device.

paelzer, 2024-04-06:
@ChicoCybergoth
Did you check on a schematic if your button is connected to a regular GPIO pin? If it is a one button device and the button is connected to a pin labled EN or RESET you can't use it as it will trigger a reset. In this case you need to connect a separate button to any of the available GPIO pins. The header file for your device in the src/drivers folder must be changed as well with the GPIO you selected accordingly. If you are working with the bin files I don't know what GPIO is selected for the one you use. But in the original header file of the source for esp32 devkit it's GPIO 0.

ChicoCybergoth, 2024-04-07:
@paelzer I dont have the schematic, but I will explain my experience.
It has 2 buttons, EN and Boot.
With EN button I simply cant get any function.
With Boot button pressed for 5 seconds I can get the device full resetted.

@sany3001 the button function I can get is the above.
What I was trying was change the wallet without fully resetting the device.
It is a minor issue, but, would be a very good function.

19201003080114, 2024-04-07:
Could it be that your EN button is not working?

ChicoCybergoth, 2024-04-07:
@19201003080114
I have tested with 10 different modules, all behave the same.

sany3001, 2024-04-07:
Just for info: On my board I have button RST and BOOT.

paelzer, 2024-04-07:
@ChicoCybergoth
Is it only "Hold 5 second..." function which is not working or do you also have no reaction on the one click and double click.

Is it this board we're talk about?
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/hw-reference/esp32/get-started-devkitc.html

paelzer, 2024-04-07:
@sany3001
How does it behave for you?
I found one of these ESP32 boards with EN and Boot button. For me it also makes full reset if I hold boot button long and power up. So I need to configure everything again not just wallet.
Is it different for you, and if so, which bin do you use? 

Boris

sany3001, 2024-04-07:
I will check it, please be patient, I will do that on Wednesday.

sany3001, 2024-04-10:
My board name is ESP32-S3-DevKitC-1 v1.0.
As I have mentioned I have buttons BOOT and RESET. If I keep pressed BOOT it inters boot mode when powered up. What happens with RESET hold when powered up I do not know, but in any of cases AP doesn't start so I cannot change configuration.

I believe that solution would be to add either button or just wire to ground some free pin - it needs to be either configured in firmware or maybe there is such a pin already. I would like to have a look on that. However I am not familiar with code base and I cannot promise some short term solution. But I would like to try. Hope I will find some time for that.

paelzer, 2024-04-10:
@sany3001
Thanks a lot for your test. I will prepare the code to check how it behaves with a different pin and come back to you when done. Guess somewhat around Saturday.

Boris

ChicoCybergoth, 2024-04-13:
@paelzer It looks the same as your link.
I bought the exact one in the "Build Tutorial" list.
Thank you.

paelzer, 2024-04-17:
Hello... Took longer than I promised... 

By defining pin 0 as "button pin 1" in the source code and then, after compiling and flashing the board (with PLATFORMIO) the Boot labeled button works for me as it described in the readme.md. 
It can switch through the different screens by single press, rotate screen by double press and clear configuration by long pressing >5 sec.

Also defining another button, I tested with GPIO 26, worked fine of course. 

The board I used was the above mentioned ESP32 c3 Devkit. 

With the bin files I wasn't successful, maybe they were created with different pin defines, I don't know, I didn't bother with them, platformio is more suitable and offers more flexibility for me. 

Boris

sany3001, 2024-04-18:
Will you create PR? Will you add small circuit diagram?

paelzer, 2024-04-21:
> Will you create PR? Will you add small circuit diagram?

Hi Sany, good evening

Do you know which of the files from platformio after compiling are the needed bin files to flash to the device with the web tool?

I will make some schematic draft next days and post here. 

Boris

sany3001, 2024-04-23:
Hi Boris,
I am sorry, I am not Nerd Miner developer, I cannot help now really.

ChicoCybergoth, 2024-04-27:
A small update.
I have got a LILYGO T-Display S3.
I could use the configuration mode normally by pressing the top button then powering the board (need to hold about 4 seconds after powering it up, the screen changes confirming), also, using the reset button instead of powering it works too (with device powered, hold the top button, click reset, wait about 4 seconds, then release the top button).

Maybe the problem is only with the ESP32-WROOM-32 firmware.

closed 2024-06-30 by BitMaker-hub

stefi01, 2025-02-15:
I had the same problem and came across a video that had the same problem so thgey used a modified nerd miner bin file which does work, try this and the file is in his description [https://www.youtube.com/watch?v=gGPa_K47aL0](url)

ChicoCybergoth, 2025-02-15:
@stefi01 If I understood correctly, he was having trouble uploading the program onto the board (flashing).
What I'm reporting here is the impossibility of entering configuration mode, which happens after successfully uploading the program, and then doing the initial configuration procedure as he did in the video.
The configuration mode I'm referring to is when the person has already done the initial configuration and wants to change some information, such as the BTC wallet or the name, or the Wi-Fi network, which isn't working. The only way I've found is to completely reset the device, which requires completely reconfiguring it by filling in all the information again.
It's a small problem, but I thought it would be interesting to fix it.

stefi01, 2025-02-15:
the reason I used his video was that....
when I tried, the four files that are supposed to be uploaded, I did the
upload but got a message saying it did not look like a config file or
something similar and failed to boot after, I used another method which did
upload the files and let me enter the credentials via a web page on my pc
but after a restart it would not connect to the router, I then came across
that guys video and used his single bin file and all the troubles I had
went away, would connect to wifi and would enter the config mode and page
and is also nerdminer v2, just one file needed, I can now get in config or
reset any time I need to with no problems at all.

GelzarZakcrakious, 2025-03-25:
i encountered the same problem on ESPDuino32/Wemos D1 R32, it had only 1 button and i flashed the ESP32 WROOM firmware