BitMaker-hub/NerdMiner_v2 #378: Bitcoin USD price incorrect & lags behind

maaalex, 2024-04-12:
for the last week the price updates seem to not work and price is not updated - at least for an hour or so… worked fine before, running 1.6.3.

What's the source/API used for the price and what's the update interval?

19201003080114, 2024-04-12:
Check here. It shows the "last_trade_price", not "price_24h"
https://github.com/BitMaker-hub/NerdMiner_v2/blob/master/src/monitor.cpp
Also I think Nerdminer only checks every hour, but update interval from api.blockchain.com is unknown.

ChimpRocket, 2024-04-22:
Hi guys,

Are there any plans to use a btc price api that updates in realtime? The main reason I got a nerd miner (today) was to get a mini display for the price but the blockchain.com "last_trade_price" hasn't changed all afternoon. 

Here's a list of API's with open auth [List of Crypto Market Data APIs](https://mixedanalytics.com/knowledge-base/top-free-crypto-apis/).

The coinbase one seems to update in realtime [https://api.coinbase.com/v2/prices/BTC-USD/spot](https://api.coinbase.com/v2/prices/BTC-USD/spot)
`{"data":{"amount":"66413.775","base":"BTC","currency":"USD"}}`
 
Cheers,
JS

ChimpRocket, 2024-04-23:
Further to my comment above.  I complied my own firmware with these changes as a test and to see if I could do it. 

The price updates every 1 minute and the dollar sign is moved as a prefix to the price.

monitor.h
```
//API BTC price
#define getBTCAPI "https://api.coinbase.com/v2/prices/BTC-USD/spot"
#define UPDATE_BTC_min   1
```

monitor.cpp
```
String getBTCprice(void){
    
    if((mBTCUpdate == 0) || (millis() - mBTCUpdate > UPDATE_BTC_min * 60 * 1000)){
    
        if (WiFi.status() != WL_CONNECTED) return "$" + String(bitcoin_price);
        
        HTTPClient http;
        try {
        http.begin(getBTCAPI);
        int httpCode = http.GET();

        if (httpCode == HTTP_CODE_OK) {
            String payload = http.getString();

            DynamicJsonDocument doc(1024);
            deserializeJson(doc, payload);
            if (doc.containsKey("data") && doc["data"].containsKey("amount")) bitcoin_price = doc["data"]["amount"];

            doc.clear();

            mBTCUpdate = millis();
        }
        
        http.end();
        } catch(...) {
          http.end();
        }
    }
  
  return "$" + String(bitcoin_price);
}
```

maaalex, 2024-04-23:
@ChimpRocket that's great, thanks for the effort! This should be merged into master.

ChimpRocket, 2024-04-24:
@19201003080114 @maaalex 
It's a bit of a hack of esp23_2432s028r.cpp but this is what my update BTC price screen looks like. 
[image]

notsureofagoodname, 2024-05-16:
+1, I bought a nerd miner as a price proxy clock. Seeing it painfully out of date isn't great. Would appreciate all prices are updated per minute/few etc.

yonijuerga, 2024-06-10:
> Further to my comment above. I complied my own firmware with these changes as a test and to see if I could do it.
>
> The price updates every 1 minute and the dollar sign is moved as a prefix to the [...]

ok, how we flash the firmware with this changes?

arcsin3x, 2024-06-25:
Do U have some free api？

BitMaker-hub, 2024-06-30:
New 1.7.0 version should improve this, can you try?

vagnerac, 2024-12-06:
BTC price does not update in the 1.7.0 version.

smharm13, 2024-12-20:
Just wanted to comment that I also made the recommended changes identified here and I its working for me now as well updating off the Coinbase price as well as the $ moving to be located in the correct place. Had never used visual studio before so this was tons over learnings for how to edit files and compile from source.