BitMaker-hub/NerdMiner_v2 #380: Black screen with backlight on

SocrasPt, 2024-04-15:
My lilygo t display s3 did the factory boot ok with text displayed on the screen. Once i flashed with the flasher.bitronics.store or even with the esptool with the provided .bin files. the display no longer displays image.

I uploaded a test screen "screendetection.bin" in other project and it worked just fine, perfect image and numbers displayed.

I tried so many times expecting to be a error but my mines doesnt display any image with this firmware. Although i can access the AP and enter WIFI configs.

PedroBalaMG, 2024-04-22:
Estou com o mesmo problema ;(

BitMaker-hub, 2024-06-30:
Can you try to flash again with new version 1.7.0? 
It improves the screen startup on TDisplayS3

SocrasPt, 2024-07-02:
I dindn't update for a while but, it had been mining without screen for a while. But a while ago the both miners did a self reset, i woke up to see them at 0 timer 0 hash. I restarted them both, and magicaly the "faulty" one started to display image too. Not sure what happened but, it is working nonstop for 70days. I can switch off screen and turn it on when ever i want. I'll reconsider updating firmware. Thank you

chnyong, 2024-12-17:
Hi. I am having the same issue. Nerdminer T-Display-S3 screen is black but the backlight is on. Is it this that i should update?
[image]

sany3001, 2024-12-17:
Try version 1.6.3.

sany3001, 2024-12-17:
If hashrate is zero (on my 5 pieces I see it quite often last days) it can be connectivity issue. Try to log data on serial port (connect via usb-c to PC).

chnyong, 2024-12-18:
> Hi. I am having the same issue. Nerdminer T-Display-S3 screen is black but the backlight is on. Is it this the correct firmware that i should select update? ![image](https://private-user-images.githubusercontent.com/42427808/396559013-560acc40-fc0b-4d4b-acdc-c905d9615dd5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ0OTUyMDksIm5iZiI6MTczNDQ5NDkwOSwicGF0aCI6Ii80MjQyNzgwOC8zOTY1NTkwMTMtNTYwYWNjNDAtZmMwYi00ZDRiLWFjZGMtYzkwNWQ5NjE1ZGQ1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE4VDA0MDgyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNhOGVkYWE3ZjcxMzYwNDk4MzAwODhkZmE4MjY5ZTJhNTE5YTZmOTgwZWZlNDUyODc3MmI4ZjA5NDdlY2E5ZGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.1p9sfZWAg5fsKYus-KczKKiCwVBCO8yBHeAtZ8qf4BY)

I just tried this. The screen wont show anything too.

sany3001, 2024-12-18:
There is another issue where guy provides manual how to flash device using ESP tool. Was working for me with 1.7.0. Maybe worth trying?

SocrasPt, 2024-12-18:
I actually never updated the firmware... Both miners are working trough usb ups for 240 days. I just touch botttom button for disabeling screen. Not quite sure what happened but the faulty just started to work for no reason.