BitMaker-hub/NerdMiner_v2 #385: ESP32-S3-DevKitC-1 support

danrodri88, 2024-04-21:
Godd day to all.

Is there a way to support this board, ESP32-S3-DevKitC-1? I have 2 o these, I tried flashing the software, but to no avail.
Thanks in advance.
https://pt.aliexpress.com/item/1005006002965361.html?spm=a2g0o.order_list.order_list_main.23.3d84caa4qeBv1o&gatewayAdapt=glo2bra

julio-matarranz, 2024-04-21:
Dear, you can try to flash with "bin/bin ESP32-S3-devKitv1"

danrodri88, 2024-04-21:
I already tried that, in first place, and devkit. And even tried others. None work. I can put in boot mode, flash all ok, but then, no AP on wifi list.

julio-matarranz, 2024-04-21:
It should work with mentioned firmware because I have a board like yours and it worked. Maybe you need to erase all before flashing.

danrodri88, 2024-04-21:
I did all of that and more. And nothing.

sany3001, 2024-04-23:
I am using such a board successfully. I recommend to review flashing procedure.
Post here what you did.

danrodri88, 2024-04-26:
I did it the same way I did with my lillygo TTgo 1.14, I cleaned the software it came with, then flashed the nerminer software.  The only difference with this board is that I had to put it in download/boot mode before doing either the wipe or the nerminer soft flsh. Is that how you did it too?

Pazuzo, 2024-04-28:
Change Flash Address of bootloader to 0x0000

danrodri88, 2024-04-28:
> Change Flash Address of bootloader to 0x0000

You´re the guy. Thanks my dear, it really worked. All the best to you.

closed 2024-04-28 by danrodri88