BitMaker-hub/NerdMiner_v2 #386: ESP32 S3 problems on upload

sercruz10, 2024-04-21:
https://a.aliexpress.com/_Evoxbs7
This board is suport by https://github.com/BitMaker-hub/NerdMiner_v2?
Wich bin should pick.
When boot say always image can not load.
Thanks

julio-matarranz, 2024-04-21:
Dear,
Have you tried to flash with "bin/bin ESP32-S3-mini-weact" ? same RGB led pin so it could be working.

sercruz10, 2024-04-21:
invalid header: ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048839 other error now with bin/bin ESP32-S3-mini-weact

19201003080114, 2024-04-23:
Try this one:
https://github.com/BitMaker-hub/NerdMiner_v2/tree/master/bin/bin%20ESP32-C3-super-mini

sercruz10, 2024-04-23:
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x1001
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: Wi-Fi,BLE
Crystal is 40MHz
MAC: 24:58:7c:dc:20:dc
Uploading stub...
Running stub...
Stub running...
but same header error 
maybe i not have the knowledge to do that.
this is a great peace of work

19201003080114, 2024-04-30:
> esp32-c3 ,s2 ,s3 all BIN flies are failed so do I

Make sure you have the correct BIN files for your device
Check that you have the correct address for each
Use proper USB-C data cables
Check that your device has the proper USB-UART driver. There are many different types used for different ESP32

tiger0007t, 2024-05-11:
https://aliexpress.ru/item/1005006027386159.html?spm=a2g2w.orderdetail.0.0.1e924aa68vKtBz&sku_id=12000037955975571

Please tell me the esp32-s3 wrom (N16R8) chips arrived with 16 megabytes of memory, which firmware is suitable for nerdminer?  I tried different ones through the online bootloader does not work

closed 2024-06-18 by BitMaker-hub