BitMaker-hub/NerdMiner_v2 #391: Dead screen, power light on

Claus-13, 2024-04-28:
Screen shows nothing, have tried different chargers even though it worked fine before, it was mining normally for several days then it just stopped showing anything on screen.

Does it have a factory reset? a wipe everything reset?

Cannot install new firmware, not recognized by pc.

sany3001, 2024-04-28:
If that device is not recognized by PC, I would double check if I have started bootloader mode.

Claus-13, 2024-04-28:
I've started in bootloader mode by pressing boot button while connecting with USB.

sany3001, 2024-04-28:
I believe there will be no USB device recognized. But you shall see new COM port. Please check it.

Claus-13, 2024-04-29:
Re-Re-checked, no new COM port. :(

sany3001, 2024-05-01:
It would be a bit surprise for me, but maybe HW is dead? Can you get one more? If both will be working at the end, then you can buy ESP32-2432S028R and collect mining effort on one aggregator. I have in total 7 miners (1x ESP32-2432S028R, 1x WROOM, 5x TTGO S3) and it is really nice looking. :-)

PSLLSP, 2024-05-22:
I have a case when LILYGO T-DONGLE S3 was overheated and it is "dead" now. I can talk to it with `esptool`, I can flash it but it doesn't start to mine or initial AP. I am not sure what is wrong but I assume some part of the chip is dead. When I try to watch serial console, I see bootloader sends several messages and then device disconnects, like it do a reset and the cycle repeats.

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cebfe
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
-- serial console is disconnected at this  point, I assume the device was reset and cycle starts again...
```

It would be nice to add some thermal protection to NerdMiner code, to suspend hashing when temperature of the chip is too high...

Claus-13, 2024-05-29:
I think this might be the case, except I cannot talk to it. It burned, so the little cooling fan I've seen on many miners, might not be a bad idea.

Thanks!

closed 2024-06-30 by BitMaker-hub