BitMaker-hub/NerdMiner_v2 #406: Bootloop Espressif32 6.7.0

julio-matarranz, 2024-05-20:
The 6.7.0 has been released 6 days ago (cf. https://registry.platformio.org/platforms/platformio/espressif32) and it causes bootloop on firmwares I tried to build with. I downgraded to 6.6.0 and no more bootloops so there is an incompatibility with 6.7.0.

BitMaker-hub, 2024-05-25:
Ouch, I'ven updated it yet. good to know

julio-matarranz, 2024-05-28:
> Ouch, I'ven updated it yet. good to know

should be releated to that : 
https://github.com/Bodmer/TFT_eSPI/issues/3284

BitMaker-hub, 2024-06-30:
@julio-matarranz do you think it affects if we just indicate the current TFT_eSPI version?

julio-matarranz, 2024-06-30:
> @julio-matarranz do you think it affects if we just indicate the current TFT_eSPI version?

Better stay with 6.6.0 for now https://github.com/Bodmer/TFT_eSPI/issues/3351

julio-matarranz, 2024-07-03:
> How do i downgrade Espressif to 6.6.0?

It's already done in platformio.ini