BitMaker-hub/NerdMiner_v2 #425: Web Dashboard after configuration

jamesturnernz, 2024-06-07:
Hi, 

We received a NerdMiner flashed chipset with no screen.  It's challenging for us to purchase and receive the screen.

Was hoping it had a small http server on it that we could connect to via a remote device to share with people on a bigger screen.

Would it be difficult to add a this feature?  
If it's a resource drain, maybe make it an option to enable during initial setup, as we don't mind it running in our case, but understand why others would.

Purely educational use on our side.
Thank you.
James

Cr4zyT1zy, 2024-06-10:
I have developed a little interface software to monitoring the nerdminer work.
You can found it on https://www.ltsoftware.it/NerdViewer.html.
Required Nerminer connect to USB and windows S.O.
Freeware software, beta version.

Cr4zyT1zy, 2024-06-10:
I have develop a little software freeware to display NerdMiner work.
It is available at this link https://www.ltsoftware.it/NerdViewer.html. 
Only for window.
Beta version.