BitMaker-hub/NerdMiner_v2 #442: 1.14 Inch TFT Display ESP32-S3 Development Board with PH2.02P Power Port for Arduino MCU Module USB Type-C ESP32 S3

GmrsKissimmee, 2024-07-02:
Hola, estoy intentando subir el FW a esta placa y no hace nada, se queda la pantalla en negro y solo una luz Vvrde intermitente rapida funcionado.

[image]
[image]
![IMG_20240630_185645](https://github.co

https://github.com/BitMaker-hub/NerdMiner_v2/assets/155043316/d87dcd59-fb73-4d06-bcc1-7ff3181995ec

m/BitMaker-hub/NerdMiner_v2/assets/155043316/40dae936-6734-4a06-91ea-90091cb9e86c)

Qs315490, 2024-09-04:
He looks similar to my device, it would be nice to have pin definitions to be sure.You can also try the firmware from my fork repository Adafruit_Feather_ESP32S3_TFT

GQster, 2026-07-29:
hey Qs315490, I cant find you Adafruit_Feather_ESP32S3_TFT repo. do you still have it?