BitMaker-hub/NerdMiner_v2 #444: output of sha256d seems wrong

betterpony, 2024-07-03:
On branch master，in file mining.cpp，I log the ‘header64’ and it's sha256d output 'hash'，it's seems that only the  4-bytes of the hash string changed each sha256d turn，the 'a5a5a5...' ，maybe 'hash[32]' not initiate I  think,  as blow：
[image]

Then：
I jump to the implement of ‘nerd_sha256d’ ，end of this function，I see something as blew:
[image]
What's the purpose here，why return when doubleHash[31] !=0 || doubleHash[30] !=0，it's not a Standard sha256d output。

Look forward to your reply

BitMaker-hub, 2024-07-03:
Hello!!

You made me think a little on this function and see it can be even improved a little bit the hashrate, Thanks! haha

Printing every iteration doesn't make sense because we are only considering those shares who have at least 2 double words with zeros. This was done because any share below this difficulty will be really low (even the actual ones too, hehe) 

Soo... This was done just to be faster on each iteration of nonces.
Then, if you want to print every share transmitted to pool just print these ones that returns true (so they are a 16bit share)

betterpony, 2024-07-05:
Thanks 

I have figured it out before your reply via read your logic carefully ，it's a great idea to burst the hash rate I think.

By the way，something I didn't understand about the **difficult from pool** and **default_difficult = 1e-4** in Macro definition:

1. Job from “public-pool.io”  is solo mining job or pool mining job?

2.If it's a solo job，why we always push **‘mining.suggest_difficulty’**  = 1e-4 to pool , I'm not sure if the following description is correct? In solo mining job, we should search for a suitable nonce based on the difficulty value specified in the solo job, such that the hash value of the block header + nonce is less than the difficulty target in the job. This difficulty target should be the BTC network-wide difficulty target。So I think the pool may not accept we suggestion（=.=）

3.If it's a pool job,  why pool didn't reject we 32-bits share ?  It's seems that if there 32-bits zero hash exist，you just try to submit the nonce，but the nonce is not suitable for the pool difficult。Is there are some simplify method for nonce share? just like your 32-bits 0 is enough（haha..）

Things above made me confuse some days($.$)

Thanks again my bro.

BitMaker-hub, 2024-07-13:
1- It's solo mining, but you allways prefer some feedback from pool to see it's working. So, in fact not providing shares until the good one could work too.
2/3- The suggest difficulty was done to try to let NerdMiner be accepted by other pools. Because the current NerdMiner difficulty is too low. Unfortunatelly it didn't work. Even stratum is different on every pool who accepts it. There are some who support. NerdMiner is giving very low hashrate, even trying to submit only at pool difficulty it  will not work.

betterpony, 2024-07-27:
Thanks my friend，
Other issues I find here,
The function in src/utils.cpp as below:

**bool checkValid(unsigned char* hash, unsigned char* target)** {
  bool valid = true;
  unsigned char diff_target[32];
  memcpy(diff_target, &target, 32);
  //convert target to little endian for comparison
  reverse_bytes(diff_target, 32);

  for(uint8_t i=31; i>=0; i--) {
    if(hash[i] > diff_target[i]) {
      valid = false;
      break;
    }
  }

  #ifdef DEBUG_MINING
  if (valid) {
    Serial.print("\tvalid : ");
    for (size_t i = 0; i < 32; i++)
        Serial.printf("%02x ", hash[i]);
    Serial.println();
  }
  #endif
  return valid;
}

1. memcpy(diff_target, &target, 32); here，the parameter **target** input is a uint8_t pointer at the begin of the function，but why you take a "&" here? I think you mean to copy a local array from **target** into **diff_target** to do some comparison next，but a “&” here may cause a doule pointer issue I think.

2.If issue **1** fixed，assume there are 2 little-end array input as below:

hash[32]   = 94850e7451d1c649e5fa131a727a49e5fa131a7e3a6e02000000000000000000
target[32] = 00000000000000000000000000000000000000003a6e03000000000000000000
[image]
I think it's a valid hash ，because 0x02<0x03，0x02 is the highest non-zero number in little-end format，so hash < target here。

after  your  **reverse_bytes(diff_target, 32);** , 
the diff_target[32] = 000000000000000000036e3a0000000000000000000000000000000000000000

So , that means we just need compare array like below:
hash[32]          = 94850e7451d1c649e5fa131a727a49e5fa131a7e3a6e02000000000000000000
diff_target[32] = 00000000000000000000000000000000000000003a6e03000000000000000000
[image]

Obvious，
  **for(uint8_t i=31; i>=0; i--) {
    if(hash[i] > diff_target[i]) {
      valid = false;
      break;
    }
  }**
  
  From index  = 31 to start the compare, how could we reach our expectation?  parameter input to the function **hash** and **target** in little-end as ablove , obvious hash < target，but via the compare here , we can't get the correct result I think.  as we all know that hash < target，means we hit a block。

golden-guy, 2024-08-01:
Hello, good catch with the memcpy call! Indeed this should use the pointer value instead. Other than that, the logic in checkValid is fine in my opinion. You need to switch endianess for diff_target to be able to compare the hash against the target.
You can check the output of each bytearray when checkValid is called, I traced a log with these:
22:02:46.767 >  hash : 13 0c 5e b2 6f 76 b6 d5 c0 e4 8e bb cf 64 22 a0 3d 37 34 89 2e 9e bf 3b 14 22 3b 7d 6e f2 00 00 
22:02:46.768 >  target : 00 00 00 00 00 00 00 00 00 03 1a be 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
22:02:46.770 >  diff_target : 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 be 1a 03 00 00 00 00 00 00 00 00 00
Note: For sake of easier logging, I disabled the check for 32bit share in mining.cpp, so that checkValid got called whenever a 16bit share was found.

Huseyin1985, 2026-11-05:
Measurement that may help pin this down, plus a correlation I could not explain mechanically.

**Setup:** ESP32-2432S028R (`HARDWARE_SHA265` enabled), V1.8.3, solo on public-pool.io, ~348 KH/s.

### False-positive rate

I reconstructed the block header from the raw stratum traffic (coinbase → merkle → 80-byte header → double-SHA256) and compared it with what the firmware reported for every share it submitted over a 3.5 h window:

| | |
|---|---|
| shares submitted | 473 |
| genuinely valid on independent reconstruction | **1** |
| false-positive rate | **99.79%** |
| pool-accepted shares that day | ~2 — consistent with the real rate |
| expected genuine shares at 348 KH/s | 1.02 per 3.5 h |

The reconstruction is trustworthy: for one share it reproduced the firmware's logged hash **exactly** (job `8cd15b7`, nonce `da79423d`, diff 1.787740496), and the intermediate merkle root matched the firmware's own `merkle sha` line. `diff_from_target()` / `le256todouble()` are correct and I did not touch them — the problem is the hash fed into them.

No false *negatives* were observed (1.02 expected, 1 found and submitted), but a single window is weak evidence for that; I would **not** yet claim the early-exit filter never misses.

### Correlation with failed TLS handshakes

The pool-stats endpoint `public-pool.io:40557` began requiring TLS 1.3, which Arduino core 2.0.x mbedTLS cannot negotiate, so every request failed with `MBEDTLS_ERR_SSL_FATAL_ALERT_MESSAGE` (-0x7780). The error branch of `getPoolData()` does not advance `mPoolUpdate`, and `printPoolData()` gates on the same variable, so retries ran at ~41/min instead of the intended 1/min. In that period:

- **Day level:** failed handshakes and submitted shares rose together, Pearson r = 0.89.
- **Within a single day** (removes the long-term trend): at 07:00 failed handshakes rose **7.5×** (449 → 3345/h) and submitted shares rose **4.6×** (18.7 → 85.8/h) while the **job rate stayed flat** (~110/h), i.e. hashrate unchanged. Within-day r = **0.894**.
- **After removing that one request**, submissions over 4 h went to **0** (a naive linear extrapolation of the fit predicted ~33; genuine expectation was 1.2). Poisson: P(0 | 33) ≈ 5e-15.

### What this does *not* show

The device still performs ~1.5 **successful** HTTPS handshakes per minute (`api.coingecko.com` every 1 min, `mempool.space` every 2 min — both accept TLS 1.2). That is ~360 handshakes in the same 4 h window, and they produced **zero** false positives.

So "the miner and mbedTLS contend for the hardware SHA peripheral", which is what I first assumed, is **not supported** — plain peripheral sharing would have shown up here too. The correlation is specifically with **failing** handshakes.

### Hypotheses I cannot separate without instrumenting the firmware

1. An aborted handshake leaves the SHA peripheral, or its lock, in a state the miner then reads.
2. Heap churn from ~41 failed connections/min disturbs the midstate / `bake` buffers.
3. CPU starvation from the retry storm races the job handoff (`s_working_current_job_id`).

I have no evidence favouring any of these, and I am not claiming the TLS path is the only source of false positives — only that removing it took the submission rate to the genuine rate in this one 4 h window.

Happy to share the header-reconstruction verifier (reads the serial log and rebuilds headers) if that would help anyone reproduce the measurement.