BitMaker-hub/NerdMiner_v2 #449: Lilygo T-Display S3 Amoled V2 (H713) No Display

beiyer838, 2024-07-16:
Hi, 

I flashed a new Lilygo T-Display S3 Amoled V2 (H713) with Nerdminer2 1.6.3 LilyGo T-Display Amoled on https://flasher.bitronics.store/.
The display remain blank while I am able to access the wifiAP for configuration.

Attaching a screenshot of the chosen firmware.  Any idea what could be wrong?
[image]

julio-matarranz, 2024-07-19:
Dear,
Can you try GitHub firmwares on the web flasher ? There is AMOLED and AMOLED touch. The difference is that touch has an io38 switch for the screen power. I didn't see any indication about that for the AMOLED V2.0.

julio-matarranz, 2024-08-26:
https://github.com/BitMaker-hub/NerdMiner_v2/discussions/464

closed 2024-08-26 by julio-matarranz