BitMaker-hub/NerdMiner_v2 #452: nerdminer-prerelease-v1.7.0 how to run it

lv333ming, 2024-07-19:
I have already burned the firmware but I don't see the program running. The screen is black. My device is LILYGO TDisplay S3

julio-matarranz, 2024-07-19:
Dear,
What model is it ? Regular or AMOLED ? With or without touch screen ?
Did you use the web flasher ?

lv333ming, 2024-07-19:
> Dear, What model is it ? Regular or AMOLED ? With or without touch screen ? Did you use the web flasher ?

The model is LILYGO TDisplay S3, ordinary TFT ST7789, this is the burning tool flash_download_tool_3.9.5  

Did you use the web flasher ?  yes

julio-matarranz, 2024-07-19:
From the readme : https://flasher.bitronics.store/

lv333ming, 2024-07-19:
> From the readme : https://flasher.bitronics.store/

I want to run version 1.7.0 to see

NeoTrace82, 2024-08-29:
tried several versions on https://flasher.bitronics.store/ using the NerdMiner-Github repository, didnt have any luck

appreciate any push into the right direction

leighklotz, 2024-10-04:
I "Successfully flashed nerdminer-prerelease-V1.7.0_NerminerV2" which is nerdminer-prerelease-V1.7.0/NerminerV2_bootloader.bin

Notice the firmware version is spelled "Ner" instead of "Nerd" in the https://flasher.bitronics.store/  Nerdminer-Github tab.

Prieto78, 2024-10-22:
> I "Successfully flashed nerdminer-prerelease-V1.7.0_NerminerV2" which is nerdminer-prerelease-V1.7.0/NerminerV2_bootloader.bin
> 
> Notice the firmware version is spelled "Ner" instead of "Nerd" in the https://flasher.bitronics.store/ Nerdminer-Github tab.

Excellent, i didn't know which one on the list I had to choose and i already have it updated.

TylonHH, 2024-10-29:
So I tried to flash `nerdminer-prerelease-V1.7.0_NerdminerV2` on my T-Disyplay-S3... No luck.

TylonHH, 2024-10-30:
This worked for me: https://github.com/BitMaker-hub/NerdMiner_v2/issues/517

dionys28, 2024-12-17:
I am having some issues setting up the WiFi. When I try to connect to the NerdMiner AP, I can't access it because I don't know the password.

TylonHH, 2024-12-18:
RTFM

PASS: MineYourCoins

Herty76, 2025-02-02:
Hello, 1.7.0 doesn't work on mine.
I don't have a hash on several pools.
Display shows 78 KH/s.
Pools used:
pool.nerdminers.org
pool.pyblock.xyz
eusolo.ckpool.org

public pool io is currently down.

Kind regards
Mark