BitMaker-hub/NerdMiner_v2 #456: Password for FW 1.7 and the settings is kept even after a flash.

Yan-Leduc, 2024-08-02:
I am new to NerdMiner and GitHub😅. I am having a problem with NerdMiner V2. I was trying to solve a problem with the WiFi not connecting. After I updated to 1.7 the password provided doesn't work. I can change the AP, but the settings for the wallet stays the same. Even after I re-flashed the NerdMiner. And I removed the feature to Save the statistics to flash memory but it comes back. What's is the correct way to reset to zero the configuration?
Update after 10 minutes I was able to connect to the NerdMiner... Weird 🤔. But It came back to the old wallet.
Thanks

TylonHH, 2024-10-29:
I also could connect to the AP. So you just waited and somehow the password `MineYourCoins` is accepted?

TylonHH, 2024-10-30:
Try this: https://github.com/BitMaker-hub/NerdMiner_v2/issues/517