BitMaker-hub/NerdMiner_v2 #459: Heltec vision master T190 claim 118kH/s base on esp32 ？？？？

betterpony, 2024-08-12:
I found a board called A, and they claim that the hash rate of the ESP32 can reach 118K. I want to know how this is achieved，pool accept it's work ，
[image]

Is there still untapped potential in the ESP32? I find their [github](https://github.com/NMminer1024/NMMiner) and [purchase link](https://www.aliexpress.com/item/1005007449552504.html)，but they only push a firmware file on github and really cheap，all of this looks real.

BitMaker-hub, 2024-08-12:
Hi! wow, I've just got surprised, hehe. 
Something to consider, really hehe

Unfortunatelly they didn't opensource the firmware, even copied almost everything

betterpony, 2024-08-12:
I was also very shocked when I first saw it. I suspect that this hash rate is fabricated.

BitMaker-hub, 2024-08-12:
I don't really know, I'm very curious. In fact they says exprimed sha256d into the limits inside the processor. This also was done  using  almost assembler.
I know there are more things to do, but don't know if they do it

JohnTaipei, 2024-08-18:
esp32-s3 + st7789 1.9"

JohnTaipei, 2024-08-26:
I am trying NMMiner V0.2.10

JohnTaipei, 2024-08-27:
it works
113KH/s
https://youtu.be/zlHMf11gVq8?si=BVC3KG2avags0nD7

HarukiToreda, 2024-09-08:
It works, I have it also using the Heltec V3 with it too at 119kh/s
[image]
[image]

HarukiToreda, 2024-09-08:
I got the firmware here
https://github.com/NMminer1024/NMMiner

BitMaker-hub, 2024-09-08:
YEs, this works but is not opensource. They just copied firmware improved it but just keeped it as a license scheme.
No more to say

closed 2024-09-08 by BitMaker-hub