BitMaker-hub/NerdMiner_v2 #466: connect to unmineable.com

isaack0815, 2024-08-25:
Hello everyone,
I have a small question.
Is it possible for me to connect the Nerdminer2 to the Sha256 pool from unmineable.com?
If so, which pool address do I have to use and what does the address have to look like?
No connection can be established with the following pool address and wallet address:
stratum+ssl://rx.unmineable.com
port: 443
ATOM:cosmos12eqyggqgskmj2t30k3hth9wj5jq4huc9qpk83y.252_nerd1
Thanks in advance for the help

Roalkege, 2024-08-28:
Hey,
you want to mine randomx. You need (I think) more than 4GB RAM and does not use sha256.
It is important that you use a sha256 coin.

Uefi1, 2025-01-07:
Hello try to get BCH on https://pool.bitsyminer.com/ I successfully connected my esp32 with NerdMiner firmware there !