BitMaker-hub/NerdMiner_v2 #468: MH ET LIVE ESP32 MiniKIT V1,1

haruzept, 2024-08-28:
Hello,

does a MH ET LIVE ESP32 MiniKIT V1,1 work with this project?

AliExpress (https://www.aliexpress.com/item/32815738795.html?spm=a2g0o.order_list.order_list_main.15.62e65c5fhM8sBf&gatewayAdapt=glo2deu) 
Color: D1 MINI ESP32

Thanks

murphy667, 2024-08-29:
thats a nice tiny board. try: https://github.com/BitMaker-hub/NerdMiner_v2/issues/467

murphy667, 2024-08-29:
i can´t see it on the pictures...are there buttons for reset and boot?

haruzept, 2024-08-29:
> i can´t see it on the pictures...are there buttons for reset and boot?

There is only one button for reset. It will starts automatically, if it has power.

murphy667, 2024-08-29:
> > i can´t see it on the pictures...are there buttons for reset and boot?
> 
> There is only one button for reset. It will starts automatically, if it has power.

This is just my thought, but I think you need to figure out how to put your device into flash mode.