BitMaker-hub/NerdMiner_v2 #471: The server goes down.

adisorin, 2024-09-04:
Server keeps going down, miners keep going down, what's going on? "public pool". The values ​​are deleted. He always takes it from the beginning!

adisorin, 2024-09-07:
288 -> >>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
10:05:28.288 -> >>> Valid blocks: 0
10:05:28.288 -> >>> Block templates: 0
10:05:28.288 -> >>> Best difficulty: 0.000
10:05:28.288 -> >>> 32Bit shares: 0
10:05:28.288 -> >>> Temperature: 39
10:05:28.288 -> >>> Total MHashes: 0
10:05:28.288 -> >>> Time mining: 0  00:03:11
10:05:28.288 -> >>> [i] Miner: newJob>false / inRun>false) - Client: connected>false / subscribed>false / wificonnected>true

adisorin, 2024-09-07:
The miners fell suddenly. From the ESP32C3 family can no longer connect to the server, why?
From 53351 they reached 26085 active online miners.
[image]

BitMaker-hub, 2024-09-08:
Pool is currently being improved, so for now you shoud wait

closed 2024-09-08 by BitMaker-hub