BitMaker-hub/NerdMiner_v2 #477: LILYGO T-Display S3 - no success flashing to 1.6.4/1.7.0

sany3001, 2024-09-06:
I am able to flash my new device to 1.6.3. It starts well. When trying 1.6.4 / 1.7.0 I see blank screen only. Is it known problem?

GeeBeeCryptos, 2024-09-08:
Same problem with my T-Display S3 !
[image]

sany3001, 2024-09-08:
See title for my device: LILYGO T-Display S3

BitMaker-hub, 2024-09-08:
hey, not a known issue. very strange. 
This is just the most common device. So should be good :(
I will check this

julio-matarranz, 2024-09-08:
Maybe it's a new device like AMOLED V2 ?

julio-matarranz, 2024-09-08:
What does the monitoring says when your board is plugged ?

GeeBeeCryptos, 2024-09-10:
New ??? Had it, since at least 9 months. And works like a charme on 1.6.3
Thank you

sany3001, 2024-09-10:
My old miner are HW 1_2. Here is boot log from HW VL2:

[image]

sany3001, 2024-10-20:
Same problem with builds 1.6.4/1.7.0 from GitHub with an old device.

TylonHH, 2024-10-29:
Wich model do you select in the prerelease?

TylonHH, 2024-10-30:
Try this: https://github.com/BitMaker-hub/NerdMiner_v2/issues/517

soenkegissel, 2025-02-08:
**Lilygo T-Display S3**

I can confirm not being able to flash 1.7.0 firmware with non of the firmwares available at https://flasher.bitronics.store/. 
Just black screen after flashing.

But 1.6.4 **NerdminerV2** firmware is working with my Lilygo T-Display S3. I flashed with https://flasher.bitronics.store/

TylonHH, 2025-02-08:
You tried my workaround?

soenkegissel, 2025-02-08:
Yes I did. Not no success. Same behavior. Not able to flash 1.7.0.
But I am satisfied with 1.6.4 if it keeps running.